Entware/Entware #94: Package request : ngrok/ngrok-c

fbigun, 2018-07-06:
Package: ngrok or ngrok-c
Exposing a local server behind a NAT or firewall to the internet.

Repo for ngrok: https://github.com/inconshreveable/ngrok
Repo for ngrok-c: https://github.com/dosgo/ngrok-c

Pagekit in repo is obviously not suitable for embedded devices

Pagekite has c language implementation libpagekite: https://github.com/pagekite/libpagekite
Ngrok also has the implementation of c language ngrokc

ryzhovau, 2018-07-06:
Did you tried `pagekite` from our repo?

fbigun, 2018-07-09:
I have tried using pagekite. He can run and use, but it is too resource intensive.

However, he is based on an application running in python and is not well suited for running on low-configuration routes.

Ngrokc is compiled as a client for c and is suitable for long-term operation on the router. And you can use the unofficial server by modifying the service information in the configuration on your own server, which is more free.

ryzhovau, 2018-09-21:
[Proven as working](https://github.com/nu1ee/asuswrt-entware-ng-ngrokc). Ok then.

closed 2018-09-21 by ryzhovau