Entware/Entware #95: Security: Package signatures and/or HTTPS

Ajedi32, 2018-07-08:
I noticed while installing Entware that the install script (in my case, <http://bin.entware.net/armv7sf-k3.2/installer/generic.sh>) seems to download all Entware's files over plain, unencrypted http, with no signature checks as far as I can tell.

This means that any attacker who can get between me and bin.entware.net (via a [BGP hijack](https://www.internetsociety.org/blog/2018/04/amazons-route-53-bgp-hijack/), upstream router compromise, or some other means) while I'm installing entware (or any entware packages?) can own my router.

I realize that fixing this is probably more complicated than just replacing every occurance of "http:" with "https:" in the install scripts, since Entware has to support a lot of environments with very outdated TLS support. (My own router's built-in wget implementation can't even seem to access https://bin.entware.net/, though http://bin.entware.net/ works fine.) However, when you're downloading arbitrary code over the internet and immediately executing it I think it's pretty important to have _some_ kind of integrity guarantees in place.

Would you consider implementing some kind of integrity check on files downloaded by the entware installer and opkg?

ryzhovau, 2018-07-08:
We can't provide HTTPS in basic installation because of reason you already know, but you can set up HTTPS connection by following [this manual](https://github.com/Entware/Entware/wiki/Using-HTTPS-for-opkg).

closed 2018-07-08 by ryzhovau

Ajedi32, 2018-07-08:
Good to know there's a way to enable HTTPS for package installs.

Are you sure there's no way to get integrity checks for the initial install though? Even if getting HTTPS to work would be too complex, couldn't you embed file hashes in the install script and have the user transfer the script to the device with scp or something?