Entware/Entware #97: fork #2 failed: 12 (Cannot allocate memory)

diman82, 2018-07-11:
When running on Entware (Asus rt-ac86u - armv8):
pyload --daemon --configdir=/opt/var/pyload

I get the following error message:
fork #2 failed: 12 (Cannot allocate memory)

How to debug this issue? What is wrong? 

P. S Already opened an issue [for](https://github.com/pyload/pyload/issues/3199) pyload, they refered me to you.

ryzhovau, 2018-07-11:
Try to add swap file first. Looks like `pyload` consumes too much RAM on start up.

diman82, 2018-07-11:
@ryzhovau Can you please refer me how this should be achieved? Yoy mean swap partition?

ryzhovau, 2018-07-11:
> Can you please refer me how this should be achieved?

Depends on current firmware. I'm not ready to provide personal support, sorry.

diman82, 2018-07-12:
Adding a swap partition worked. 
You may close this issue.

closed 2018-07-12 by zyxmon