Entware/Entware #98: unbound 1.7.3

underd0se, 2018-07-11:
Unbound version in the Entware repo is 1.6.7-1 and dated back to Oct 2017. 

`# opkg info unbound
Package: unbound
Version: 1.6.7-1
`

Newest version which has lots of fixes for critical bugs and vulnerabilities, according to their [SVN](https://unbound.nlnetlabs.nl/svn/) is 1.7.3. 

Can you update the Entware repo with the latest version for each platform? 

Thanks in advance.

zyxmon, 2018-07-11:
The current version of unbound is 1.7.1 in Entware. @underd0se - probably you are speaking about Entware-ng, that is not maintained now. We will update unbount with next openwrt sync - https://github.com/openwrt/packages/blob/master/net/unbound/Makefile

ryzhovau, 2018-09-11:
Done. Binaires will arrive soon.

closed 2018-09-11 by ryzhovau