Entware/Entware #104: rtorrent

KamikazeePL, 2018-07-15:
I also get #https://github.com/Entware/Entware/issues/96  `The futex facility returned an unexpected error code.Aborted`
when I try to install rtorrent easy install. Error shows when script try to start rtorrent.

KamikazeePL, 2018-07-15:
When I made clean instalation from http://bin.entware.net/mipselsf-k3.4/installer/alternative.sh transmission works without error.

KamikazeePL, 2018-07-16:
Clean instalation for Transmission works also normally.
`Clean install 

 wget http://bin.entware.net/mipselsf-k3.4/installer/generic.sh
Connecting to bin.entware.net (81.4.123.217:80)
generic.sh           100% |*******************************|  2200   0:00:00 ETA
root@DD-WRT:/opt# sh generic.sh
Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
Info: Opkg package manager deployment...
Connecting to bin.entware.net (81.4.123.217:80)
opkg                 100% |*******************************|   143k  0:00:00 ETA
Connecting to bin.entware.net (81.4.123.217:80)
opkg.conf            100% |*******************************|   150   0:00:00 ETA
Connecting to bin.entware.net (81.4.123.217:80)
ld-2.27.so           100% |*******************************|   155k  0:00:00 ETA
Connecting to bin.entware.net (81.4.123.217:80)
libc-2.27.so         100% |*******************************|  1613k  0:00:00 ETA
Connecting to bin.entware.net (81.4.123.217:80)
libgcc_s.so.1        100% |*******************************| 95288   0:00:00 ETA
Connecting to bin.entware.net (81.4.123.217:80)
libpthread-2.27.so   100% |*******************************|   116k  0:00:00 ETA
Info: Basic packages installation...
Downloading http://bin.entware.net/mipselsf-k3.4/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
Installing entware-opt (227000-3) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/entware-opt_227000-3_all.ipk
Installing libc (2.27-8) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libc_2.27-8_mipsel-3.4.ipk
Installing libgcc (7.3.0-8) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libgcc_7.3.0-8_mipsel-3.4.ipk
Installing libssp (7.3.0-8) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libssp_7.3.0-8_mipsel-3.4.ipk
Installing librt (2.27-8) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/librt_2.27-8_mipsel-3.4.ipk
Installing libpthread (2.27-8) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libpthread_2.27-8_mipsel-3.4.ip                                                                                                                                                             k
Installing libstdcpp (7.3.0-8) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libstdcpp_7.3.0-8_mipsel-3.4.ip                                                                                                                                                             k
Installing entware-release (1.0-2) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/entware-release_1.0-2_all.ipk
Installing zoneinfo-asia (2018e-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/zoneinfo-asia_2018e-1_mipsel-3.4.ipk
Installing zoneinfo-europe (2018e-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/zoneinfo-europe_2018e-1_mipsel-3.4.ipk
Installing findutils (4.6.0-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/findutils_4.6.0-1_mipsel-3.4.ipk
Installing terminfo (6.1-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/terminfo_6.1-1_mipsel-3.4.ipk
Installing locales (2.27-8) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/locales_2.27-8_mipsel-3.4.ipk
Installing grep (2.26-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/grep_2.26-1_mipsel-3.4.ipk
Installing libpcre (8.41-2) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libpcre_8.41-2_mipsel-3.4.ipk
Installing opkg (2011-04-08-9c97d5ec-17b) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/opkg_2011-04-08-9c97d5ec-17b_mipsel-3.4.ipk
Installing entware-upgrade (1.0-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/entware-upgrade_1.0-1_all.ipk
Configuring libgcc.
Configuring libc.
Configuring libssp.
Configuring libpthread.
Configuring librt.
Configuring terminfo.
Configuring libpcre.
Configuring grep.
Configuring locales.
Entware uses separate locale-archive file independent from main system
Creating locale archive - /opt/usr/lib/locale/locale-archive
[... 63 lines cut ...]
[2018-07-16 17:00:39.539] Cache Maximum cache size set to 2.00 MiB (128 blocks) (cache.c:261)
[2018-07-16 17:00:39.540] RPC Server Adding address to whitelist: 127.0.0.1 (rpc-server.c:971)
[2018-07-16 17:00:39.540] RPC Server Adding address to whitelist:  (rpc-server.c:971)
[2018-07-16 17:00:39.540] RPC Server Adding address to whitelist: 192.168.1.* (rpc-server.c:971)
[2018-07-16 17:00:39.540] RPC Server Serving RPC and Web requests on port 127.0.0.1:9091/transmission/ (rpc-server.c:1213)
[2018-07-16 17:00:39.540] RPC Server Started listening on 0.0.0.0:9091 (rpc-server.c:853)
[2018-07-16 17:00:39.540] RPC Server Whitelist enabled (rpc-server.c:1217)
[2018-07-16 17:00:39.540] Bound socket 15 to port 2015 on 0.0.0.0 (net.c:388)
[2018-07-16 17:00:39.541] Port Forwarding Stopped (port-forwarding.c:180)
[2018-07-16 17:00:39.541] UDP Failed to set receive buffer: requested 4194304, got 344064 (tr-udp.c:84)
[2018-07-16 17:00:39.541] UDP Please add the line "net.core.rmem_max = 4194304" to /etc/sysctl.conf (tr-udp.c:89)
[2018-07-16 17:00:39.541] UDP Failed to set send buffer: requested 1048576, got 344064 (tr-udp.c:95)
[2018-07-16 17:00:39.541] UDP Please add the line "net.core.wmem_max = 1048576" to /etc/sysctl.conf (tr-udp.c:100)
[2018-07-16 17:00:39.541] DHT Initializing DHT (tr-dht.c:278)
[2018-07-16 17:00:39.541] DHT Reusing old id (tr-dht.c:307)
[... cut ...]
[2018-07-16 17:00:39.542] LPD Local Peer Discovery initialised (tr-lpd.c:354)
[2018-07-16 17:00:39.542] DHT Bootstrapping from 49 IPv4 nodes (tr-dht.c:156)
[2018-07-16 17:00:39.542] Using settings from "/opt/etc/transmission/" (daemon.c:528)
[2018-07-16 17:00:39.542] Saved "/opt/etc/transmission/settings.json" (variant.c:1266)
[2018-07-16 17:00:39.542] Watching "/opt/etc/transmission/watchdir" for new .torrent files (daemon.c:573)

`
**I think it was my mistake I did not restarted router between updates.**

zyxmon, 2018-07-16:
Some time this helps - https://github.com/Entware/Entware/wiki#note-about-packages-upgrade
Or this (reinstalling all libs) - 
`opkg list-installed|grep -v "libc "|sed 's/ - .*$//'|grep lib|grep -v libpthread|grep -v libgcc|xargs -n 5 opkg --force-reinstall install
`

closed 2018-07-16 by zyxmon

KamikazeePL, 2018-07-16:
I run this command and installed deluge for test but I get same error 
`The futex facility returned an unexpected error code.Aborted`  it's good for me  that transmission and aria2 normally

KamikazeePL, 2018-07-16:
Python -> The futex facility returned an unexpected error code.Aborted

KamikazeePL, 2018-07-16:
I going back Entware-NG because Transmission still get error when I open web interface.

zyxmon, 2018-07-17:
It looks like a bug in dd-wrt kernel for mipsel. I guess it is broadcom.

KamikazeePL, 2018-07-17:
```
uname -a
Linux DD-WRT 3.2.102 #4098 Fri Jun 29 16:45:04 CEST 2018 mips DD-WRT
```

KamikazeePL, 2018-07-17:
Is there any to bug this?

zyxmon, 2018-07-18:
mip and mipsel are different. uname for mipsel  returns mips - this is normal.
Nothing wrong with the kernel version. The kernel itself may have bugs because of closed source broadcom drivers.

KostyaChel, 2018-07-28:
I got exactly same issue and found that the kernel don't have **futex** enabled. 

Steps to reproduce the issue:
1. Install **opkg** from http://bin.entware.net/mipselsf-k3.4/installer/ by using **generic.sh** script;
2. Install at **lighttpd** and **strace**;
3. Execute `strace lighttpd -f /opt/etc/lighttpd/lighttpd.conf` and look up at output:
> set_robust_list(0x77987560, 12)         = -1 ENOSYS (Function not implemented)
> futex(0x779020b4, FUTEX_WAKE_PRIVATE, 2147483647) = -1 ENOSYS (Function not implemented)

I guess that the kernel 3.2.102 in DD-WRT for ASUS RN-N13U B1 firmware was built using config **.config_rt3052_usb** (see [sorce](https://svn.dd-wrt.com/browser/src/linux/universal/linux-3.2)) where **# CONFIG_FUTEX is not set*.
I not sure this is actually the source which using for DD-WRT kernel, but I'm sure either the software (rtorrrents, transmission, lighttpd etc) should not use the **futex** or the kernel should support it.

Do somebody knows whether it possible to replace kernel (**vmlinuz** file) in built firmware from another firmware, where **futex** enabled. I found that nearest firmware config **.config_rt3052_8366rb** has futex enabled.