Entware/Entware #107: Update prosody to 0.10.2

Rooby69, 2018-07-19:
Package request
---
{Describe killer feature of package}
- sometmes tprosody crashes
- new scurity release 0.10.2 see
http://blog.prosody.im/prosody-0-10-2-security-release/

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I successfully tested added package on my {platform}.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

Rooby69, 2018-09-13:
Please update, prosody crashes from time to time, not usable.
After some debuggig I found the following crash of prosody using gdb:

Program received signal SIGSEGV, Segmentation fault.
0x2ac9e9a0 in luaL_addlstring () from /opt/lib/liblua.so.5.1.5

this looks like that the liblua library should also be updated.

ryzhovau, 2018-09-13:
Both `prosody` and `liblua` came from upstream, so please, [send](https://github.com/openwrt/packages/issues) this issue to OpenWrt guys.

Upstream updates will came here after next sync, but I'm not sure `liblua` will be updated. LUCI WebUI depends on it and OpenWrt guys very conservative with it.

closed 2018-09-13 by ryzhovau