Entware/Entware #111: Samba - add streams_xattr module

pztrn, 2018-07-31:
This module is neccessary to create proper share for Time Machine.

Right now Samba says:

```
Error loading module '/opt/lib/samba/vfs/streams_xattr.so': /opt/lib/samba/vfs/streams_xattr.so: cannot open shared object file: No such file or directory
error probing vfs module 'streams_xattr': NT_STATUS_UNSUCCESSFUL
smbd_vfs_init: vfs_init_custom failed for streams_xattr
```

zyxmon, 2018-08-01:
What is your arch (repo). I can upload /opt/lib/samba/vfs/ folder for testing.

pztrn, 2018-08-01:
mipsel (kngiga3).

zyxmon, 2018-08-01:
I do not compile mips & mipsel packages. May be someone else will upload it. libs are compiled but not installed.

ryzhovau, 2018-08-01:
@pztrn, check out [samba-4.8.0_full.tgz](https://bin.entware.net/mipselsf-k3.4/test/samba-4.8.0_full.tgz) to find out what libraries should be packed at next build.

Waiting for feedback.

pztrn, 2018-08-01:
At least ``/opt/lib/samba/vfs/{catia,fruit,streams_xattr}.so``.

Here's Samba config that should do well as Time Machine thing:

```
[TimeMachine]
	path = /media/MACDATA/TimeMachine
	browseable = yes
	writeable = yes
	create mask = 0600
	directory mask = 0700
	spotlight = yes
	vfs objects = catia fruit streams_xattr
	fruit:aapl = yes
	fruit:time machine = yes
```

According to this, these 3 libraries are required.

pztrn, 2018-08-01:
(changed issue name to properly reflect issue's topic)

ryzhovau, 2018-09-19:
`samba4` package was heavily reworked by OpenWrt team. Looks like we need to implement additional package with `/opt/lib/samba/vfs/*` content. Stay tuned.

ryzhovau, 2018-12-27:
Synced. Now vfs modules provided as in OpenWrt.

closed 2018-12-27 by ryzhovau