Entware/Entware #112: Package Request: Wireguard

tuxflo, 2018-08-01:
### Wireguard

As described here: https://github.com/Entware/Entware-ng/issues/715
it might be a good alternative for OpenVPN or similar VPN providing tools.

Since a lot of boxes I know (Routers, NAS Systems and so on) have been updated to a more recent kernel "Entware is usually installed on hardware running 2.6 kernel." is not a thing anymore (in my opinion)
### Plattform
QNAP

zyxmon, 2018-08-02:
Wireguard requires kernel 3.10 or higher and a kernel module that cannot be built in Entware.

closed 2018-08-02 by zyxmon

tuxflo, 2018-08-03:
I did not know that Entware is bind to a specific kernel version, but other then that there is also a "userspace version" which is described here: https://www.wireguard.com/xplatform/

zyxmon, 2018-08-03:
Entware require kernel to equal or new then used in Entware toolchain.
Entware does not have kernel modules because they are specific to device and firware,

wireguard-go is written in go language that is usually to heavy for embedded devices. Userspace wireguard version will certainly be slow.

HuJK, 2019-10-03:
But no matter how slow it is, I think it still faster than openvpn. Openvpn are userspace implement, too.
I thank we can use https://github.com/cloudflare/boringtun instead, which is written in Rust.

rtorrentuser, 2019-12-15:
> Entware require kernel to equal or higher then used in Entware toolchain.
> Entware does not have kernel modules because they are specific to device and firware,
> 
> wireguard-go is written in go language that is usually to heavy for embedded devices. Userspace wireguard version will certainly be slow.

Some people report that wireguard-go is running fine on ASUS routers with old kernel, much faster then OpenVPN and consuming less resources...
https://www.snbforums.com/threads/wireguard-implementation.59784/page-2#post-521697

tuxflo, 2019-12-15:
There is also [boringun](https://github.com/cloudflare/boringtun) which is a userspace implementation of the wireguard protocol, written in rust.

rtorrentuser, 2019-12-18:
> Some people report that wireguard-go is running fine on ASUS routers with old kernel, much faster then OpenVPN and consuming less resources...
> https://www.snbforums.com/threads/wireguard-implementation.59784/page-2#post-521697

Finally, I decided to try wireguard-go on my RT-AC68U router with latest firmware from Merlin...
I compiled wireguard-go using go_nohf_1.12.10-1_armv7-2.6.ipk. Also I compiled 'wg' tool from the main repository and slightly adjusted original scripts, in particular, to set txqueuelen to 1000 (500 by default) for the virtual interface, created by  wireguard-go (after that warnings 'interface ... wants to queue packet' have disappeared from router log). 
Wireguard has been running smoothly for several days now on my router in client mode. Speeds are comparable to those of OpenVPN and are even slightly higher... And it is not too resource-hungry...
Swap is needed, but is not used aggressively - mainly when transferring large files FROM router over wireguard.
If someone has the same router, feel free to try it yourself - you can find everything necessary including binaries in the attached file (remember to adjust 'wg0.conf'!). 'bash' and 'libmnl' packets should be installed! Firewall rules in my case:

-A POSTROUTING -o wg0 -j MASQUERADE
-A INPUT -m state --state NEW -i wg0 -j ACCEPT
-A FORWARD -m state --state NEW -i wg0 -j ACCEPT

I think, having 'wireguard-go' in entware repository is a good idea, since it's the only option to use wireguard on routers based on old kernels.
[WireGard_RT-AC68U.tar.gz](https://github.com/Entware/Entware/files/3977836/WireGard_RT-AC68U.tar.gz)

herbi3, 2020-02-19:
> > Some people report that wireguard-go is running fine on ASUS routers with old kernel, much faster then OpenVPN and consuming less resources...
> > https://www.snbforums.com/threads/wireguard-implementation.59784/page-2#post-521697
> 
> Finally, I decided to try wireguard-go on my RT-AC68U router with latest firmware from Merlin...
> I compiled wireguard-go using go_nohf_1.12.10-1_armv7-2.6.ipk. Also I compiled 'wg' tool from the main repository and slightly adjusted original scripts, in particular, to set txqueuelen to 1000 (500 by default) for the virtual interface, created by wireguard-go (after that warnings 'interface ... wants to queue packet' have disappeared from router log).
> Wireguard has been running smoothly for several days now on my router in client mode. Speeds are comparable to those of OpenVPN and are even slightly higher... And it is not too resource-hungry...
> Swap is needed, but is not used aggressively - mainly when transferring large files FROM router over wireguard.
> If someone has the same router, feel free to try it yourself - you can find everything necessary including binaries in the attached file (remember to adjust 'wg0.conf'!). 'bash' and 'libmnl' packets should be installed! Firewall rules in my case:
> 
> -A POSTROUTING -o wg0 -j MASQUERADE
> -A INPUT -m state --state NEW -i wg0 -j ACCEPT
> -A FORWARD -m state --state NEW -i wg0 -j ACCEPT
> 
> I think, having 'wireguard-go' in entware repository is a good idea, since it's the only option to use wireguard on routers based on old kernels.
> 
> UPD (2020-01-04):
> One more hint, that makes swap unnecessary for wireguard at least on my RT-AC68U ...
> In device/queueconstants_default.go replace:
> 
> MaxSegmentSize = (1 << 16) - 1 // largest possible UDP datagram
> PreallocatedBuffersPerPool = 0 // Disable and allow for infinite memory growth
> 
> with (taken from device/queueconstants_android.go):
> 
> MaxSegmentSize = 2200
> PreallocatedBuffersPerPool = 4096
> 
> This tweak to the wireguard-go code makes it use less RAM without visible impact on speed.
> 
> [WireGard_RT-AC68U.tar.gz](https://github.com/Entware/Entware/files/4022426/WireGard_RT-AC68U.tar.gz)

would there potentially be something available for the GT-AC5300? or RT-AC5300? it would be great to have something to work off

zyxmon, 2020-02-24:
Entware package wireguard-go with low ram patch is ready for testing. Plz post your arch to get test package.

herbi3, 2020-02-24:
ARMv7 Processor rev 0 (v7l)
Model RT-AC5300
Firmware Version 384.15
Firmware Build Sat Feb 8 18:40:05 UTC 2020 merlin@da0c8cd
Bootloader (CFE) 1.0.4.0
Driver version wl0: Apr 18 2019 10:40:26 version 10.10.69.69030 (r774262) FWID 01-1b7c8a6b

zyxmon, 2020-02-24:
@herbi3 - plz post `cat /opt/etc/entware_release`

herbi3, 2020-02-24:
release=entware
arch=arm
cpu=cortex-a9
cpu_subtype=unknown
float=soft
gcc=8.3.0
gcc_flags=-O2 -pipe -mtune=cortex-a9 -fno-caller-saves -mfloat-abi=soft
libc=glibc
libc_version=2.23

zyxmon, 2020-02-24:
I guess this is your package - http://bin.entware.net/armv7sf-k2.6/t/wireguard-go_nohf_v0.0.20200121-1_armv7-2.6.ipk
wireguard-tools is in the repo.

herbi3, 2020-02-24:
will give it a crack and report back

aikinai, 2020-09-16:
Looks like this hasn't moved in a a long time, but I'd like to give it a try if possible.

RT-AC88U
```
release=entware
arch=arm
cpu=cortex-a9
cpu_subtype=unknown
float=soft
gcc=7.4.0
gcc_flags=-O2 -pipe -march=armv7-a -mtune=cortex-a9 -fno-caller-saves -mfloat-abi=soft
libc=glibc
libc_version=2.23
```

zyxmon, 2020-09-16:
@aikinai - wireguard-go is not stable on old 2.6 kernels. It was removed.

aikinai, 2020-09-16:
Oh okay, that's too bad. Thanks for the super quick response!