Entware/Entware #120: Package request: python-libtorrent

diman82, 2018-08-08:
Package request: python-libtorrent (python implementation of the popular libtorrent library)
---
{The main goals of libtorrent are:

- to be cpu efficient
- to be memory efficient
- to be very easy to use

}

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I successfully tested added package on my {Windows 10, Ubuntu 16.04}.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

zyxmon, 2018-08-08:
Do you mean this one - https://github.com/Entware/rtndev/tree/master/rblibtorrent
It has python bindings and is used in deluge?
Adding library without executable has little sence.

ryzhovau, 2018-09-10:
No response.

closed 2018-09-10 by ryzhovau