Entware/Entware #125: Patchelf package suddenly missing

stefaang, 2018-08-30:
Package request
---
Allows patching binaries on the machine itself.

- [ x ] I have installed patchelf from entware earlier... but suddenly it's not available anymore?
- [ x ] I successfully tested added package on my x86-64 platform.

---

zyxmon, 2018-08-30:
```
# opkg list | grep patchelf
patchelf - 0.9-1 - A small utility to modify the dynamic linker and RPATH of ELF executables
# cat /opt/etc/entware_release
release=entware
arch=x86_64
cpu=
cpu_subtype=unknown
float=hard
gcc=7.3.0
gcc_flags=-O2 -pipe -fomit-frame-pointer -fno-caller-saves
libc=glibc
libc_version=2.27

```

closed 2018-08-30 by zyxmon

stefaang, 2018-08-30:
Okay my bad, I accidentally used an older entware-ng installation.
Would it make sense to add a deprecation warning mechanism in the `opkg update` command?