Entware/Entware #126: Help: /opt/bin/opkg: can't execute: Permission denied

alienx2, 2018-08-31:
i had installed entware on my router was perfect without trouble. but Im having problem with install entware on Android TV box.

> root@rk3188:/ # uname -a
> Linux localhost 3.0.36+ #175 SMP PREEMPT Thu May 22 11:44:15 CEST 2014 armv7l GNU/Linux
> root@rk3188:/ #

I chosen Kernel 2.6 bec i tried 3.1+ giving error with chmod said “Bad mode”

**Install log:**
> Connecting to bin.entware.net (81.4.123.217:80)
> opkg                 100% |********************************|  127k  0:00:00 ETA
> Connecting to bin.entware.net (81.4.123.217:80)
> opkg.conf            100% |********************************|   190  0:00:00 ETA
> Connecting to bin.entware.net (81.4.123.217:80)
> ld-2.23.so           100% |********************************|  139k  0:00:00 ETA
> Connecting to bin.entware.net (81.4.123.217:80)
> libc-2.23.so         100% |********************************| 1186k  0:00:00 ETA
> Connecting to bin.entware.net (81.4.123.217:80)
> libgcc_s.so.1        100% |********************************| 45272  0:00:00 ETA
> Connecting to bin.entware.net (81.4.123.217:80)
> libpthread-2.23.so   100% |********************************| 92640  0:00:00 ETA
> link failed Operation not permitted
> link failed Operation not permitted
> link failed Operation not permitted
> Info: Basic packages installation...
> sh: <stdin>[43]: /opt/bin/opkg: can't execute: Permission denied
> sh: <stdin>[44]: /opt/bin/opkg: can't execute: Permission denied
> sh: <stdin>[45]: /opt/bin/opkg: can't execute: Permission denied
> cp: /opt/etc/passwd.1: No such file or directory
> cp: /opt/etc/group.1: No such file or directory
> cp: /opt/etc/shells.1: No such file or directory
> Info: Congratulations!
> Info: If there are no errors above then Entware was successfully initialized.
> Info: Add /opt/bin & /opt/sbin to your PATH variable
> Info: Add '/opt/etc/init.d/rc.unslung start' to firmware startup script for Entware services to start
> 
> This is an alternative Entware installation. We recomend to install and setup Entware version of ssh server
> and use it instead of a firmware supplied one. You can install dropbear or openssh as an ssh server

/opt/bin/opkg: can't execute: Permission denied ??? anyone know?

zyxmon, 2018-08-31:
Entware installation on android is always tricky. There is no single recipe. In your case it looks like the filesystem with /opt folder does not support or alow symlinks.

alienx2, 2018-08-31:
any idea? i will try test...

zyxmon, 2018-08-31:
Just read my blog (Russian) - http://www.zyxmon.org 
You cna find some post on installing Entware on android.
More ideas here (Russian) - http://forums.zyxmon.org/viewforum.php?f=21

alienx2, 2018-09-03:
Thank you for everything you guided us there.. i found out it required linux file system like ext2,ext3,ext4 only… i was tested internal was working then external is not… i didnt know it was ext4 but i saw i wonder why its fat32 cant run it like it said permission denied. hehehe so now i did format it to ext4 was working everything smooth.

also keep in mind: **When install entware**, it need to enter ‘**su**’ superuser mode and **mount /system**

running install on android tv box same as asuswrt which kernal 2.6 armv7… 

**for who have problem with bad address (or related with resolv.conf):** need to update busybox 1.22 or higher version working fine. 
-**How to check version of busybox?** try enter ‘wget’ in terminal it will tell you first line that version of busybox.