Entware/Entware #134: libevent in PHP

darkrain88, 2018-09-14:
Package request
---
first many thank for your update all packages.

i use 'opkg update' and 'opkg upgrade'

then my Nextcloud cant run properly, PHP +mysql +nginx in my openwrt

in the /opt/lib/php most library have been updated except the following:

[image]

in my log of nextcloud

the show the libevent direct to 2.05  now its 2.18 stable

[image]
 
php version

root@Openwrt:~# php-cli -v
PHP Warning:  **PHP Startup: Unable to load dynamic library 'libevent.so' (tried:                                /opt/lib/php/libevent.so (libevent-2.0.so.5: cannot open shared object file: No                                such file or directory), /opt/lib/php/libevent.so.so (/opt/lib/php/libevent.so.s                               o: cannot open shared object file: No such file or directory)) in Unknown on lin                               e 0
PHP 7.2.9 (cli) ( NTS )**
Copyright (c) 1997-2018 The PHP Group
Zend Engine v3.2.0, Copyright (c) 1998-2018 Zend Technologies
    with Zend OPcache v7.2.9, Copyright (c) 1999-2018, by Zend Technologies

root@Openwrt:~# php-fcgi -v
**PHP Warning:  PHP Startup: Unable to load dynamic library 'libevent.so' (tried: /opt/lib/php/libevent.so (libevent-2.0.so.5: cannot open shared object file: No such file or directory), /opt/lib/php/libevent.so.so (/opt/lib/php/libevent.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0**
PHP 7.2.9 (cgi-fcgi)
Copyright (c) 1997-2018 The PHP Group
Zend Engine v3.2.0, Copyright (c) 1998-2018 Zend Technologies
    with Zend OPcache v7.2.9, Copyright (c) 1999-2018, by Zend Technologies

root@Openwrt:~# mysql -V
mysql  Ver 15.1 Distrib 10.2.17-MariaDB, for Linux (x86_64) using readline 5.1

in /opt/lib
libevent have been installed

[image]

would you please help me out of this issue?

many thanks.

{Describe killer feature of package}

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2018-09-14:
Please, reinstall php7 packages.
https://github.com/Entware/Entware/wiki#note-about-packages-upgrade

closed 2018-09-14 by ryzhovau

darkrain88, 2018-09-15:
thanks

since update many problem occurred