Entware/Entware #138: Package Update: OpenSSL 1.1.1

D1n0Bot, 2018-09-15:
https://www.openssl.org/news/openssl-1.1.1-notes.html
Most exciting is the support of TLS 1.3. Please look into it soon.

Major changes between OpenSSL 1.1.0i and OpenSSL 1.1.1 [11 Sep 2018]
Support for TLSv1.3 added (see https://wiki.openssl.org/index.php/TLS1.3 for further important information). The TLSv1.3 implementation includes:
Fully compliant implementation of RFC8446 (TLSv1.3) on by default
Early data (0-RTT)
Post-handshake authentication and key update
Middlebox Compatibility Mode
TLSv1.3 PSKs
Support for all five RFC8446 ciphersuites
RSA-PSS signature algorithms (backported to TLSv1.2)
Configurable session ticket support
Stateless server support
Rewrite of the packet construction code for "safer" packet handling
Rewrite of the extension handling code
Complete rewrite of the OpenSSL random number generator to introduce the following capabilities
The default RAND method now utilizes an AES-CTR DRBG according to NIST standard SP 800-90Ar1.
Support for multiple DRBG instances with seed chaining.
There is a public and private DRBG instance.
The DRBG instances are fork-safe.
Keep all global DRBG instances on the secure heap if it is enabled.
The public and private DRBG instance are per thread for lock free operation
Support for various new cryptographic algorithms including:
SHA3
SHA512/224 and SHA512/256
EdDSA (both Ed25519 and Ed448) including X509 and TLS support
X448 (adding to the existing X25519 support in 1.1.0)
Multi-prime RSA
SM2
SM3
SM4
SipHash
ARIA (including TLS support)
Significant Side-Channel attack security improvements
Add a new ClientHello callback to provide the ability to adjust the SSL object at an early stage.
Add 'Maximum Fragment Length' TLS extension negotiation and support
A new STORE module, which implements a uniform and URI based reader of stores that can contain keys, certificates, CRLs and numerous other objects.
Move the display of configuration data to configdata.pm.
Allow GNU style "make variables" to be used with Configure.
Claim the namespaces OSSL and OPENSSL, represented as symbol prefixes
Rewrite of devcrypto engine
Client DoS due to large DH parameter (CVE-2018-0732)
Cache timing vulnerability in RSA Key Generation (CVE-2018-0737)
Constructed ASN.1 types with a recursive definition could exceed the stack (CVE-2018-0739)
Incorrect CRYPTO_memcmp on HP-UX PA-RISC (CVE-2018-0733)
rsaz_1024_mul_avx2 overflow bug on x86_64 (CVE-2017-3738)

ryzhovau, 2018-09-15:
This packages came from [upstream](https://github.com/openwrt/openwrt/tree/master/package/libs/openssl) with no any changes. Please, [send](https://bugs.openwrt.org/) this issue to OpenWrt team and new version will arrive here with next sync.

closed 2018-09-15 by ryzhovau

D1n0Bot, 2018-09-15:
Thanks. Seems like they are already working on it... hope it will come out soon... 
https://github.com/openwrt/openwrt/pull/965

darkrain88, 2018-09-16:
good next time。nginx can support tls1.3

D1n0Bot, 2019-02-13:
Here it come... OpenSSL 1.1.1a.. woohoo...
Thanks in advance for the update. Soon.....
https://github.com/openwrt/openwrt/commit/d872d00b2f7e31b98e11e83922d1aaefc270647e