Entware/Entware #139: pixelserv-tls 2.1.2 segmentation fault

token0, 2018-09-15:
Installed via *opkg install pixelserv-tls*

My system:
```
# cat /proc/version
Linux version 3.4.113 (padavan@hms) (gcc version 4.6.4 (GCC) ) #1 SMP Sun Jun 3 14:43:52 MSK 2018

# uname -mr
3.4.113 mips

# cat /proc/cpuinfo |grep type
system type             : MediaTek MT7621 SoC

# cat /proc/cpuinfo |grep model
cpu model               : MIPS 1004Kc V2.15
cpu model               : MIPS 1004Kc V2.15
cpu model               : MIPS 1004Kc V2.15
cpu model               : MIPS 1004Kc V2.15

```

Tried init.d script (*ARGS="0.0.0.0 -p 7080 -k 7443 -z /opt/var/cache/pixelserv"*)
```
# /opt/etc/init.d/S80pixelserv-tls start
 Starting pixelserv-tls...              failed.
```

Then tried manual start:
```
# pixelserv-tls 0.0.0.0 -p 7080 -k 7443 -z /opt/var/cache/pixelserv -l 3 -f
pixelserv-tls[13742]: pixelserv-tls 2.1.2 (compiled: Sep  8 2018 20:33:38) options: 0.0.0.0 -p 7080 -k 7443 -z /opt/var/cache/pixelserv -l 3 -f
Segmentation fault
```

ryzhovau, 2018-09-15:
Can't reproduce that. Try to install Entware from scratch.
```
~ # opkg install pixelserv-tls
Installing pixelserv-tls (2.1.2-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/pixelserv-tls_2.1.2-1_mipsel-3.4.ipk
Configuring pixelserv-tls.
Place ca.crt and ca.key to /opt/var/cache/pixelserv before starting HTTPS mode.
Start options can be adjusted via /opt/etc/init.d/S80pixelserv-tls.

See 'pixelserv-tls -h' for details
~ # pixelserv-tls 0.0.0.0 -p 7080 -k 7443 -z /opt/var/cache/pixelserv -l 3 -f
pixelserv-tls[20158]: pixelserv-tls 2.1.2 (compiled: Sep  8 2018 20:33:38) options: 0.0.0.0 -p 7080 -k 7443 -z /opt/var/cache/pixelserv -l 3 -f
pixelserv-tls[20158]: cert_tlstor_init: failed to load ca.crt
pixelserv-tls[20158]: cert_tlstor_init: failed to load ca.key
pixelserv-tls[20158]: sslctx_tbl_load: /opt/var/cache/pixelserv/prefetch doesn't exist.
pixelserv-tls[20158]: Listening on :0.0.0.0:7080
pixelserv-tls[20158]: Listening on :0.0.0.0:7443
^C
~ # grep 7621 /proc/cpuinfo
system type             : MediaTek MT7621 SoC
```

token0, 2018-09-15:
Could it be GCC version used for building firmware?

closed 2018-10-03 by no-response[bot]