Entware/Entware #142: Package request: simpleproxy (aka. Simple-TCP-proxy)

lars18th, 2018-09-18:
https://github.com/vzaliva/simpleproxy

Very small TCP "redirector" with HTTP proxy tunnel support with authentication.
Similar to others like 3proxy, but more lightweight!

Please consider to add it.

ryzhovau, 2018-09-19:
What's the killer feature of `simpleproxy`? 
I see no sense to add several packages with similar purpose, sorry. There's already bunch of proxies present: `tinyproxy`, `haproxy`, `privoxy`, `redsocks`, `dante`, `srelay` and so on.

ryzhovau, 2018-09-19:
> haproxy: fat HTTP proxy.

See `mode tcp`.  So, there's at least couple of packages with similar purpose and the only `simpleproxy` killer feature is the binary size. Still not sure it will be useful, sorry.

lars18th, 2018-09-19:
Hi @ryzhovau ,

Thank you for your feedback. This project is amazing, and your work is great.

>>haproxy: fat HTTP proxy.

>See mode tcp. So, there's at least couple of packages with similar purpose and the only simpleproxy killer feature is the binary size. Still not sure it will be useful, sorry.

It's true. The `haproxy` can work as a TCP redirector (when using the reverse proxy mode). Futhermore, the well known `nginx` can do it as well. However, these projects are FAT... very fat. But, for a very simple, small and fast TCP redirection no more simple than this project exists. Even with the added killer functionality of pass through an HTTP proxy with user/pass Authentication.

Please, take note that the source of this project is very simple. It compiles in a lot of platforms. And all regular Linux distributions include it in the standalone repos. So support it can be very easy.

Futhermore, the target of the ENT-WARE project isn't small devices?
In this case, an small footprint is a must have!

Thus, please consider it. :wink: 
Thank you!

ryzhovau, 2019-04-27:
No more TCP/UPD/HTTP proxies, sorry.

Consider using one of already available.

closed 2019-04-27 by ryzhovau