Entware/Entware #145: clamav

swetoast, 2018-09-26:
Package request
clamav 
{Describe killer feature of package}
An antivirus for linux
- [ x ] No similar package already added and/or this one got some new killer feature(s),
- [ x ] Package is in active development (at least, not abandoned for 5 years or more),
- [ x ] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

swetoast, 2018-09-27:
@zyxmon @ryzhovau still working on entware ?

Dmole, 2019-04-11:
clamav is RAM heavy, so on embedded devices will not scan larger files ( just pointing out the RoI on this might be lower than expected)

swetoast, 2019-04-11:
Works just fine on my nas without issues all it takes is a bit of swap

closed 2019-05-07 by ryzhovau