Entware/Entware #152: Package request: LoggedFS

lars18th, 2018-10-04:
Package request: LoggedFS
---
{LoggedFS is a fuse-based filesystem which can log every operations that happens in it.}
- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),

---
LoggedFS - Filesystem monitoring with Fuse
https://github.com/rflament/loggedfs

In embbeded systems it's good to know how works some system processes. This package, using the FUSE filesystem can show the activity in the filesystem. The good point is that the requirements to compile it are low: only three libraries.

ryzhovau, 2018-10-05:
`inotify-tools` seems to have similar purpose. Did you tried it?

lars18th, 2018-10-05:
Hi @ryzhovau ,

Thank you! It works!

However, one key option is missing in `inotify-tools`: you can't see the process that access to the file.
I'm trying to use in conjunction with `lsof` (already inside Entware), but without success.

Regards.

zyxmon, 2018-12-16:
Package added https://github.com/Entware/rtndev/commit/461e84f552d7c1e7bef9aeedd33e603f64531144

closed 2018-12-16 by zyxmon