Entware/Entware #153: Package "znc_1.7.1-2_mipsel-3.4" appears to ship with a broken executable

StepS-, 2018-10-05:
After upgrading from Entware-3x to the new Entware repo and then upgrading from **znc_1.6.5-1_mipsel-3x** to **znc_1.7.1-2_mipsel-3.4**, the znc executable no longer works. Trying to invoke "znc" results in "not found", even if done inside the /opt/bin directory, no matter the permissions. Same issue with opkg reinstalled completely from scratch. At the same time, the executable from 1.6.5-1 works. I went through the manual process of rolling back to 1.6.5-1 with its dependencies and that worked.

NDMS 2.13.C.0.0-0, if that matters.

The-BB, 2018-10-05:
please, update the package list, reinstall znc, and try again

StepS-, 2018-10-05:
Thank you, it's working now.

closed 2018-10-05 by StepS-