Entware/Entware #155: Package request: sg3_utils

lars18th, 2018-10-07:
Package request: sg3_utils
---
{Unique low level commands for SCSI devices (or emulated, like SAT AHCI)}

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),

---
https://github.com/hreinecke/sg3_utils
  The package includes a number of utilities to allow a user
  to use the sg interface to find out low-level information
  about any SCSI device.

Inside a NAS device is a must have.

Note: OpenWRT make file for this package:
https://git.openwrt.org/?p=openwrt/svn-archive/archive.git;a=blob_plain;f=utils/sg3-utils/Makefile;hb=e90c317aa53d9b5752b91cb8b9546c92b0816ed2

closed 2019-05-07 by ryzhovau

lars18th, 2019-05-07:
Thank you @ryzhovau !

ryzhovau, 2019-05-07:
mipsel binaries are ready, x86 and other will arrive a bit later.