Entware/Entware #158: Illegal instruction

diman82, 2018-10-13:
Suddenly, I get an error of "Illegal instruction" for every Entware command I run on my Asus RT-AC86U router.
Tried to run the [Entware installation script](https://github.com/Entware/Entware/wiki/Install-on-asuswrt-merlin-firmware), but it finishes right away with the same error.
Upgarded the router to the newest RT-AC86U_384.7_0 firmware, which didn't help as well.

What could be the problem and how to fix it?

amplatfus, 2018-12-24:
Hi,

I know this is old, but could you please share how did you solved this if you solved.
Thanks!

amplatfus, 2018-12-24:
Thank you. Posted [here](https://github.com/Entware/Entware/issues/201#issuecomment-449758161) and solve it already.
Merry Christmas and Happy New year!

closed 2018-12-25 by ryzhovau