Entware/Entware #160: Package nfs-kernel-server don't work on my wd cloud home

Albaju, 2018-10-14:
Package don't work

Hello I'm having a problem with the package:
nfs-kernel-server - 2.3.2-2

Material: WD MY CLOUD HOME
Entware bin: armv7sf-k3.2

when I run S57nfs-kernel-server start, I get the following error message:

```
Starting NFS server
mount: /opt/var/lib/nfs/rpc_pipefs: unknown filesystem type 'rpc_pipefs'.
mount: /proc/fs/nfsd: unknown filesystem type 'nfsd'.
./S57nfs-kernel-server[76]: can't create /proc/fs/nfsd/max_block_size: No such file or directory
rpc.nfsd: Unable to access /proc/fs/nfsd errno 2 (No such file or directory).
Please try, as root, 'mount -t nfsd nfsd /proc/fs/nfsd' and then restart rpc.nfsd to correct the problem
rpc.nfsd: unable to set any sockets for nfsd
```

Should I miss something?
Thank you for your help.

zyxmon, 2018-10-14:
You need apropriate kernel modules for kernel server that should be provided by your device firmware.
If you do not have nfs kernel modules - use unfs package that provides user space implementation.

closed 2018-10-14 by zyxmon

Albaju, 2018-10-14:
hi Zyxmon,

I made a new installation of Entware.
I installed the package unfs3

unfortunately, the service does not start:
/opt/etc/init.d # ./S56unfsd check
  Checking unfsd ... dead.

/opt/etc/init.d # ./S56unfsd start
  Starting unfsd ... failed.

An idea

ryzhovau, 2018-10-15:
Please, configure any package __before__ start.

PS We do not provide personal support here, sorry.

Albaju, 2018-10-16:
Hi ryzhovau,
I configured the export file in /opt/etc/exports ike this:

`/data/wd/diskVolume0/samba/share/NFS_ADMSYSFILE 10.10.10.0/23(ro,all_squash,insecure)`

There is another file to configure because I can not find any information anywhere?

I also, this error message when I want to install PORTMAP:
```
/system/bin # opkg install portmap
Unknown package 'portmap'.
Collected errors:
 * opkg_install_cmd: Cannot install package portmap.
```
This one is no longer in the repo http://bin.entware.net/armv7sf-k3.2
Thank you for your help.

zyxmon, 2018-10-16:
portmap was recently replaced with rpcbind.

Albaju, 2018-10-19:
Hello,
It's been over a week since I tried to run the nfs server unsuccessfully.
I did some research on the internet, a lot of tests but nothing works.

Can you tell me if the result of the following command prevents the unfsd service from starting:

```
/opt/etc/init.d # rpcbind -adfw
rpcbind: using '/opt/var/run/rpcbind/rpcbind.xdr' startup file
rpcbind: using '/opt/var/run/rpcbind/portmap.xdr' startup file
can't get local ip4 address: Servname not supported for ai_socktype
can't get local ip6 address: Servname not supported for ai_socktype

```
and if so, what can I do?

```
/opt/etc/init.d # rpcinfo
   program version netid     address                service    owner
    100000    4    local     /opt/var/run/rpcbind.sock -          superuser
    100000    3    local     /opt/var/run/rpcbind.sock -          superuser
```
```
/opt/etc/init.d # ps | grep rpc*
  510 root         0 SW   [rpc-1]
  511 root         0 SW   [rpc-3]
14687 root      2448 S    rpcbind -f -w
14943 root      4380 S    grep rpc*
```
```
iptables -F
/opt/etc/init.d # iptables -nvL
Chain INPUT (policy ACCEPT 166 packets, 13480 bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination

Chain OUTPUT (policy ACCEPT 121 packets, 12252 bytes)
 pkts bytes target     prot opt in     out     source               destination
```

```
UNFS3 unfsd 0.9.23 (C) 2009, Pascal Schmidt <unfs3-server@ewetel.net>
Cannot register service: RPC: Unable to receive; errno = Connection refused
unable to register (NFS3_PROGRAM, NFS_V3, udp).
```

```
/opt/etc # cat services
sunrpc 111/tcp rpcbind # SUN Remote Procedure Call
sunrpc 111/udp rpcbind # SUN Remote Procedure Call
```

Thank

zyxmon, 2018-10-20:
This is a place to report bugs and request new packages.
We do not provide this sort of support here. We do not have your WD device. Probably you should solve your issue on some WD forums.

PS I run rpcbind on my router without errors and warnings. So I guess something is missing on your WD.

Albaju, 2018-10-25:
I finally managed to run the NFS server.
Everything is [here](https://community.wd.com/t/install-entware-on-wd-my-cloud-home-ssh-access-nfs-server-opkg-install-packages/228591)

https://community.wd.com/t/install-entware-on-wd-my-cloud-home-ssh-access-nfs-server-opkg-install-packages/228591

Thanks for your help.