Entware/Entware #162: fork of entware-ng

maurerr, 2018-10-20:
no Package request
---
{Describe killer feature of package}

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {platform} and get a feedback about it.

---
Hi guys,

As I've noticed you archived the entware-ng repos and the mipselsf is not maintained anymore I've taken the time to recompile most of the newest packages using entware-ng toolchain.
I didn'nt had time to create patches/push changes for the packages - just manual editing and compiling so far
My question to you - is it ok to use the "entware-ng2" name for the fork or should I find something else?

Thanks,
M

ryzhovau, 2018-10-20:
Please, tell us why you still on entware-ng?
I suspect you've got some kernel 2.6 based firmware, which is preventing to use new kernel 3.4 based repo.

maurerr, 2018-10-20:
yes, I plan to support only mipselsf for asuswrt-john's fork and tomato - the last 2 kernel-2.6.22 firmwares still maintained

ryzhovau, 2018-10-20:
We just discussing possible new `mipsel-k2.6` arch for our repo, so give us a timeout for couple of days. 

I won't return to ancient k2.6.22 which causes too much changes and limiting number of packages available, so I prefer to pick k3.6.36 as in `armv7sf-k2.6` arch.

maurerr, 2018-10-20:
unfortunately broadcom sdk for mipsel (for asuswrt and tomato) is limited to 2.6.22 kernel :(

ryzhovau, 2018-10-20:
At least, it closer than k3.4 and I hope it will run smoothly. You may test `mipsel-k2.6` and give us a feedback in future.

Also, a huge part of k2.6 MIPSEL devices based on MTK SoCs: MT7628, MT7620, MT7621. We want them to "upgrade" to Entware too.

maurerr, 2018-10-20:
entware-backports name its fine by my.
i'll register a domain now