Entware/Entware #165: samba 4 and files 2+ GB

fixed77, 2018-10-26:
Hello

samba-server 4.8.4-1 not support large files larger then 2GB.
in samba-3.6 is ok.

zyxmon, 2018-10-26:
It may be upstream (openwrt) bug. We use the same Makefile as openwrt now.

gadget1999, 2018-10-28:
I tested the version on OpenWrt, it seems to be working with files over 2GB. The only difference with OpenWrt version installs to root folder (with overlay), the Entware version installs on /opt folder.

d25fe0be, 2018-10-30:
Experiencing the same issue here.

Log says:

```
smbd_dirptr_lanman2_mode_fn: Couldn't stat [...] (Value too large for defined data type)
```

It seems that smbd is using a type that is too small to handle large files. 

`smbd -b` confirms this:

```
Type sizes:
   sizeof(char):         1
   sizeof(int):          4
   sizeof(long):         4
   sizeof(long long):    8
   sizeof(uint8_t):      1
   sizeof(uint16_t):     2
   sizeof(uint32_t):     4
   sizeof(short):        2
   sizeof(void*):        4
   sizeof(size_t):       4
   sizeof(off_t):        4  <-- !!!
   sizeof(ino_t):        4
   sizeof(dev_t):        8
```

Unfortunately I could not get a OpenWrt version of samba4 to see if it works. But if the Makefile we're using is the same as OpenWrt and OpenWrt's samba4 does work, I would guess it's the toolchain / building env. which caused the difference. (@gadget1999 Would you mind checking the size of `off_t` in OpenWrt's samba4 by running `smbd -b`?)

Presumably we should somehow (`-D_FILE_OFFSET_BITS=64`, `-D_LARGEFILE64_SOURCE=1`, or something) instruct samba4 to use a 64bit `off_t` if its `configure` does not get it right for any reason.

gadget1999, 2018-10-31:
Here is the output of smbd -b for the OpenWrt version. Looks it's compiled with 64 bit option.

Type sizes:
   sizeof(char):         1
   sizeof(int):          4
   sizeof(long):         4
   sizeof(long long):    8
   sizeof(uint8_t):      1
   sizeof(uint16_t):     2
   sizeof(uint32_t):     4
   sizeof(short):        2
   sizeof(void*):        4
   sizeof(size_t):       4
   sizeof(off_t):        8
   sizeof(ino_t):        8
   sizeof(dev_t):        8

fixed77, 2018-10-31:
smbd -b on mi-r3g router (mipsel):
`Type sizes:
   sizeof(char):         1
   sizeof(int):          4
   sizeof(long):         4
   sizeof(long long):    8
   sizeof(uint8_t):      1
   sizeof(uint16_t):     2
   sizeof(uint32_t):     4
   sizeof(short):        2
   sizeof(void*):        4
   sizeof(size_t):       4
   sizeof(off_t):        4
   sizeof(ino_t):        4
   sizeof(dev_t):        8
`

amiklosi, 2018-11-21:
having the same issue on  DD-WRT v3.0-r37305 std (R7000). Any chance you could try to recompile?

ryzhovau, 2018-12-26:
No samba4 became a part of OpenWrt feed. Just checked `smbd -b` out, seems to be ok.

closed 2018-12-26 by ryzhovau