Entware/Entware #172: Package request: exempi

kfeeeeee, 2018-11-12:
Package request
---
"exempi is a port of Adobe XMP SDK to work on UNIX". This would be especially useful for all OCR-related topics.

- [x ] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {platform} and get a feedback about it.

Wasn't able to compile it by myself yet, since I am stuck with with some boost issues -.-

zyxmon, 2018-11-13:
We have tesseract OCR in Entware. Have you tried it?

kfeeeeee, 2018-11-13:
It might be possible to compile on my armv7 but I think libboost-dev is missing.

kfeeeeee, 2018-11-15:
So, I tried to compile example after but got an

> ../libtool: line 2639: func__fatal_error: command not found

It seems to be a bug in libtool that was fixed in > 2.4.6 - 26. I am not an expert on this so I am really not sure what the problems is.

kfeeeeee, 2018-11-16:
Yes, I found this as well. However, I am really stuck now. I did build an older version of libtool as well but still cannot build exempi. It seems to compile well till the end but fails just when it's packaging the libs with this libtool error...:(

zyxmon, 2018-11-16:
It is much easier. You needed to modify one line in text file. We cannot give such sort of support here.

kfeeeeee, 2018-11-16:
okay, so, again I am stuck with 

> libtool:   error: unsupported hardcode properties See the libtool documentation for more information. Fatal configuration error.

And this point, I'll give up tbh. Might be just impossible to compile it on this architecture dont know. Thanks for your help. You might close this issue.

closed 2018-11-16 by ryzhovau