Entware/Entware #174: Please update netdata to v1.11

sinancetinkaya, 2018-11-15:
Package update request
---
Developer's notice https://www.facebook.com/groups/251560641854558/permalink/764757460534871/
> We just released netdata v1.11. This release fixes a number of known vulnerabilities, so please update asap.

ryzhovau, 2018-12-24:
Done, binaries will arrive soon.

closed 2018-12-24 by ryzhovau

camkidd, 2018-12-31:
Netdata seems to have been updated to `1.11.0-1`, but program version still says `1.10.1_rolling`, so not sure if it is actually new version.

```$ opkg list netdata
netdata - 1.11.0-1 - netdata is a highly optimized Linux daemon providing real-time ...
$ netdata -V
netdata 1.10.1_rolling```

zyxmon, 2019-01-01:
@camkidd - this is a bug in version reporting ftom netdata sources
https://github.com/netdata/netdata/issues/4572