Entware/Entware #175: [ISSUE] [TRANSMISSION][SYSCTL.D]

Danielperalta07, 2018-11-16:
Hello, I'm having the following issue:

**Packages:** 
[transmission-daemon-openssl - 2.94-2]
[transmission-web - 2.94-2]

**Error:**
[UDP Failed to set receive buffer: requested 4194304, got 1966080 (tr-udp.c:84)]

**Solution:**

1 - [editing /opt/etc/sysctl.d/20-transmission.conf]

modifying the values to:

`net.core.rmem_max = 16777216 `
`net.core.wmem_max = 4194304`

2 - aplying changes:

 sysctl -p /opt/etc/sysctl.d/

3 - restarting the transmission's deamon

works fine!!

[**NEW PROBLEM**]

This manual process, steps 2-3, are necessary on every reboot because kernel isn't applying `/opt/etc/sysctl.d/` automatically on boot. Also I cannot create a `/etc/sysctl.d/*.conf` because system automatically deletes( don't know why =/ ) created files on this directory.

SO THE MAIN PROBLEM HERE ( i think ) IS `/opt/etc/sysctl.d/` NOT BEING AUTOMATICALLY APLIED ON BOOT.

Tried some solutions found on web but, since entware is a diferent system did'nt have success.

Thanks in advance for the time to read and support!

ryzhovau, 2018-11-16:
0. That's not an issue.
1. System wide changes in a favor of one utility is not right.
2. There's no universal memory settings for every embedded devices.
3. We can't assume `sysctl` present on any embedded device.

closed 2018-11-16 by ryzhovau

Danielperalta07, 2018-11-16:
Got, thank you anyway for the time!

But, since i'm a newbie could please explain why "We can't assume sysctl present on any embedded device." ?

ryzhovau, 2018-11-17:
My bad, I presume you want to use `sysctl` from firmware, which is rarely present.

RogueScholar, 2018-11-19:
@Danielperalta07 Before I switched to using aria2, I encountered the same issue. My admittedly lazy fix was to modify the /opt/etc/init.d/S80transmission file (or whatever the relevant init script filename is, it's been a while now since I've switched) so that the PREARGS line read as follows:

`PREARGS="/bin/echo 16777216 > /proc/sys/net/core/rmem_max; /bin/echo 4194304 > /proc/sys/net/core/wmem_max"`

This has the benefit of only modifying system values when the transmission daemon is being started and not relying on an absent sysctl as an intermediary; sysctl is just a user-friendly way to spare people from remembering these /proc paths anyway.

You can read about this topic in much greater detail [here](https://www.snbforums.com/threads/torrenting-best-practices-fw-version-tweaks.20946/#post-152937), written by someone far, far smarter than me. :wink: