Entware/Entware #176: Go package doesn't work : Illegal instruction

NatoBoram, 2018-11-19:
I installed Go on my R8000 using `opkg install go` and I added `/opt/bin/go/bin` to my `$PATH`.

```bash
root@DD-WRT:~# opkg install go
Installing go (1.11-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/go_1.11-1_armv7-3.2.ipk
Configuring go.
Please add /opt/bin/go/bin to your PATH
Please set GOROOT=/opt/bin/go environment variable to use GO compiler

root@DD-WRT:~# go
Illegal instruction

root@DD-WRT:~# /opt/bin/go/bin/go
Illegal instruction
```

zyxmon, 2018-11-19:
> I installed Go on my R8000
Is it a model of you Liebherr refrigirator? Which version (architecture) of Entware it is running? Next time please be more specific!
My guess it has Broadcom arm cpu and in this case you need to install *-nohf go and packages -
`opkg install go_nohf`. _nohf stands for No Hard Float.

closed 2018-11-19 by zyxmon

NatoBoram, 2018-11-19:
Oh, sorry! It's a [Nighthawk AC3200 Tri-Band WiFi Router](https://www.netgear.com/home/products/networking/wifi-routers/R8000.aspx).

```yaml
root@DD-WRT 
----------- 
OS: Linux 4.4.159 armv7l 
Kernel: 4.4.159 
Uptime: 24 mins 
Packages: 96 (opkg) 
Shell: sh 
Terminal: /dev/pts/0 
CPU: Northstar Prototype (2) 
Memory: 53MiB / 248MiB 
```

I'm currently downloading the `go_nohf` package and I'll be back as soon as it's installed.