Entware/Entware #178: Update Packages.gz

RadixOneZero, 2018-11-19:
```
root# opkg update
Downloading http://pkg.entware.net/binaries/armv7/Packages.gz.
Updated list of available packages in /opt/var/opkg-lists/packages.
root# opkg install shadowsocks-libev
Installing shadowsocks-libev (3.0.6-1) to root...
Downloading http://pkg.entware.net/binaries/armv7/shadowsocks-libev_3.0.6-1_armv7soft.ipk.
Collected errors:
 * opkg_download: Failed to download http://pkg.entware.net/binaries/armv7/shadowsocks-libev_3.0.6-1_armv7soft.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download shadowsocks-libev. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package shadowsocks-libev.

```

zyxmon, 2018-11-19:
Your Entware-ng installation automatically upgrades to Entware. Something went wrong - or probably you have not read the instructions after running `opkg update` one of the previous times.
1. Try to run `opkg update` once more - this may swith you to Entware repo.
2. If it does not help - you must reinstall Entware. Please install Entware, not Entware-ng. Entware-ng is not supported now.

closed 2018-11-19 by zyxmon

RadixOneZero, 2018-11-19:
Thank you, the problem is solved.