Entware/Entware #179: lighttpd: 1.4.48-2 - missing openssl module

yecarrillo, 2018-11-19:
Package request
---
Please add support for openssl on lighttpd armv7 (again)

Openwrt resolved this [already](https://github.com/openwrt/packages/issues/6012)

ryzhovau, 2018-12-26:
Lighty came from OpenWrt with all it's modules, except `authn_gssapi` (Kerberos authentification). 
The issue you referenced was resolved in May 2018, at Entware was synced with OpenWrt at least twice since than. 
Could it be just a mis-configuration, no?

yecarrillo, 2018-12-31:
Thank you Alexander. Maybe packagename changed?
```
/tmp/home/root# opkg install lighttpd-mod-openssl                                                                                                              
Unknown package 'lighttpd-mod-openssl'.                                                                                                                                         
Collected errors:                                                                                                                                                               
 * opkg_install_cmd: Cannot install package lighttpd-mod-openssl.     
```

http://pkg.entware.net/binaries/armv7/Packages.html 
[image]

ryzhovau, 2018-12-31:
No any additional modules required. Just use SSL features as you planned.
You may check compile-time features by `lighttpd -V`.

closed 2018-12-31 by ryzhovau