Entware/Entware #197: Package request: extundelete

hoochlemon, 2018-12-21:
Package request
---
extundelete and the needed libraries (ext2fs library), to undelete files.

- [x ] No similar package already added and/or this one got some new killer feature(s),
- [x ] I'm ready to test added package on my {platform} and get a feedback about it.
- [x ] I'm ready to write a howto for QNAP NAS in the Forum.

Hello Entware-devs
I am happy about the new Version of Entware, it works great, thanks!
I want to recovery a Folder...., but i cant compile with gcc extundelte 0.2.4 (downloaded from http://extundelete.sourceforge.net/)-> illegal arguments and Can't find ext2fs library Errors. 
I installed libext2fs, gcc, on the QNAP-x86-64-System. But i am not sure what is needed more for extundelete.

best regards

scrobbleme, 2019-02-19:
Same here, would like to this as well.

ryzhovau, 2019-05-07:
Some users report about segfaults when they attempt to use it. I've applied Debian patches but the rest part of test is yours.

`extundelete` is abandoned for now, I want to be sure it's useful in current state. Please, provide a feedback.

closed 2019-05-07 by ryzhovau