Entware/Entware #201: rclone (1.42-1) Illegal instruction

amplatfus, 2018-12-24:
Package error
---
{Describe Package error of rclone (1.42-1)}

Hi,

I had entware-ng. I updated to entware using `opkg update; opkg upgrade` 2 times.
Everything works well. Today I installed rclone. Every command that I entered, I got illegal instruction error like below. Could somebody provide some advice?
Thank you so much!
Merry Christmas and Happy New Year!

```
user@router:/tmp/home/root# opkg install rclone
Installing rclone (1.42-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/rclone_1.42-1_armv7-2.6.ipk
Configuring rclone.
user@router:/tmp/home/root# rclone --version
Illegal instruction
user@router:/tmp/home/root# rclone --help
Illegal instruction
user@router:/tmp/home/root# uname -r
2.6.36.4brcmarm
user@router:/tmp/home/root# gdb rclone help
GNU gdb (GDB) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rclone...(no debugging symbols found)...done.
/tmp/home/root/help: No such file or directory.
(gdb) 
```

zyxmon, 2018-12-24:
In your cpu is broadcom - you need to install rcclone_nohf package
The same as here - https://github.com/Entware/Entware/issues/176
and similar issues.

closed 2018-12-24 by zyxmon

amplatfus, 2018-12-24:
Works, thank you so much. Merry Christmas and Happy New Year!

diman82, 2018-12-24:
@amplatfus You had a specific problem with rclone app, I had a global problem with everything..