Entware/Entware #203: install guide not work for ds214play (platform i686)

jheroy, 2018-12-25:
Package request
---
install guide

zyxmon, 2018-12-25:
Entware developers do not have synology nases to check Entware. We cannot help you without any logs and additional info.

closed 2018-12-25 by zyxmon

jheroy, 2018-12-25:
I use this command to install    `wget -O -
http://bin.entware.net/x64-k3.2/installer/generic.sh | /bin/sh`

/bin/sh: line 43: /opt/bin/opkg: cannot execute binary file: Exec format
error
/bin/sh: line 44: /opt/bin/opkg: cannot execute binary file: Exec format
error

zyxmon, 2018-12-25:
It looks like your kernel does not have 64 bit support. What is the output of 'uname -a'. In case your kernel is 32 bit, you can install Entware-ng. But it is not supported and updated.

jheroy, 2018-12-25:
Thanks ，its works for me.