Entware/Entware #204: Fix i2pd init script

dartraiden, 2018-12-26:
i2pd на старте не может найти требуемые сертификаты:
```
01:04:07@576/warn - Family: Can't load family certificates from /opt/var/lib/i2pd/certificates/family
01:04:07@576/warn - Reseed: Can't load reseed certificates from /opt/var/lib/i2pd/certificates/reseed
```

это совершенно логично, поскольку сертификаты лежат в /opt/etc/i2pd/certificates. 
Аналогичная проблема и с tunnels.conf, он расположен в /opt/etc/i2pd, но i2pd безуспешно будет пытаться найти его в рабочем каталоге /opt/var/lib/i2pd

Похоже, что init писался на основе OpenWrt-шного, там была такая же несостыковка, которую решили созданием симлинков при каждом запуске i2pd
https://github.com/openwrt/packages/commit/86106ecb8020645a932c237f635424198734ddf6

ryzhovau, 2018-12-26:
> Похоже, что init писался на основе OpenWrt-шного

Изначально это наш пакет, позже затащенный в OpenWrt.
Спасибо за репорт. Действительно, это наш [косяк](https://github.com/Entware/entware-packages/commit/4150fbb627b90a3cc503b19e7c37f2bcd177dd68#diff-e55a23c0fe1b9b382562440f26c5c537) в процессе проведения последней синхронизации с OpenWrt.

dartraiden, 2018-12-26:
Да, вижу, проблема с tunnels.conf, получается, решалась простым указанием пути к нему в конфиге.

dartraiden, 2018-12-26:
Если вернусь на OpenWrt, когда-нибудь, то надо будет поправить, да :D

dartraiden, 2018-12-26:
Можно где-то почитать, почему subscriptions.txt deprecated? В репозитории они пока присутсвуют
https://github.com/PurpleI2P/i2pd/blob/openssl/contrib/subscriptions.txt

ryzhovau, 2018-12-27:
https://github.com/Entware/entware-packages/commit/a5bb998912b3ea18b720dd1d36ec75ba9e9813c6

closed 2018-12-27 by ryzhovau