Entware/Entware #206: Transmission not starting since last update

HazMatt69, 2018-12-30:
[/opt/etc/init.d] # ./S88transmission start
 Starting transmission-daemon...              failed. 
[/opt/etc/init.d] #

amplatfus, 2018-12-30:
A suggestion is to uninstall and install again. On your risk.

Before please take a look here:
https://github.com/Entware/Entware/issues/205#issuecomment-449954732

Good luck!

HazMatt69, 2018-12-30:
Yeah, that's what I did but no luck :-(

amplatfus, 2018-12-30:
I am sorry. You could try again like in the link I shared.
Good luck!

HazMatt69, 2018-12-30:
As I've said, I tried that but it doesn't work... Thanks anyways

zyxmon, 2018-12-30:
@HazMatt69 - you can reinstall all libs with `opkg list-installed|grep -v "libc "|sed 's/ - .*$//'|grep lib|grep -v libpthread|grep -v libgcc|xargs -n 5 opkg --force-reinstall install`.
You can also install and use ldd to find out what is wrong with libs.

closed 2018-12-30 by zyxmon

HazMatt69, 2018-12-30:
@zyxmon Thanks man!