Entware/Entware #207: Package perl Locale::gettext missing?

der-knut, 2019-01-01:
Package request
---
{perl Locale::gettext needed for several projects, e.g. vdradmin-am.pl}

- [X] No similar package already added?,
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my {platform} and get a feedback about it.

Unable to install with CPAN even if libintl and perl dev is installed as mentioned here
https://github.com/Entware/Entware-ng/wiki/Self-installation-of-perl-modules

```
cpan[3]> install Locale::gettext
Running install for module 'Locale::gettext'
Checksum for /storage/.cpan/sources/authors/id/P/PV/PVANDRY/gettext-1.07.tar.gz ok
Scanning cache /storage/.cpan/build for sizes
............................................................................DONE
'YAML' not installed, will not store persistent state
Configuring P/PV/PVANDRY/gettext-1.07.tar.gz with Makefile.PL
checking for gettext... no
checking for gettext in -lintl... no
gettext function not found. Please install libintl at Makefile.PL line 18.
Warning: No success on command[/storage/.opt/bin/perl5.28.1 Makefile.PL]
  PVANDRY/gettext-1.07.tar.gz
  /storage/.opt/bin/perl5.28.1 Makefile.PL -- NOT OK
Failed during this command:
 PVANDRY/gettext-1.07.tar.gz                  : writemakefile NO '/storage/.opt/bin/perl5.28.1 Makefile.PL' returned status 512

```

zyxmon, 2019-01-01:
We are ware of this problem. Openwrt designed gettext and libiconv to be in separate folders. 
https://github.com/Entware/Entware/blob/master/include/nls.mk
It is probably not a problem to specify library path in your case, but he headers will be still missing, they are not packed in tar.gz.
You can try to compile lib yourself. I'll keep the issue open - may be it will be fixed in future.

der-knut, 2019-01-04:
I copied the relevant files from
[liblocale-gettext-perl](https://packages.debian.org/sid/liblocale-gettext-perl)
and this seems to work.
perl:gettext is not an unusual usage, please cosider to add a package like  
[liblocale-gettext-perl](https://packages.debian.org/sid/liblocale-gettext-perl)

ryzhovau, 2019-05-07:
We won't provide every Perl module requested, sorry. 
So we added all modules from OpenWrt and leaving the rest part up to users.

https://github.com/Entware/Entware/wiki/Self-installation-of-perl-modules

closed 2019-05-07 by ryzhovau