Entware/Entware #211: Please add gprof

l-g, 2019-01-07:
Package request
---
Profiling! I have a problem with another entware package and the developers have asked me to profile it.

https://sourceware.org/binutils/docs/gprof/index.html

- [X] No similar package already added and/or this one got some new killer feature(s),

```
# opkg find .*profil.*
# 
```
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my {platform} and get a feedback about it.

ryzhovau, 2019-05-07:
We've got `valgrind` and `orpofile`. What's the killer feature of `gprof`?
We trying to avoid to add several packages with the same purpose.

ryzhovau, 2019-05-08:
```
opkg install binutils
```

closed 2019-05-08 by ryzhovau