Entware/Entware #214: Transmission HTTPS tracker could not connect

krehman, 2019-01-09:
When connecting to private tracker with expired certificate - how can I force transmission to ignore SSL verification? I have `transmission-daemon-openssl - 2.94-2`. In the log I see:

```
Jan  8 14:10:28 transmission-daemon[22386]: web will verify tracker certs using envvar CURL_CA_BUNDLE: none (web.c:400)

Jan  8 14:10:28 transmission-daemon[22386]: web NB: this only works if you built against libcurl with openssl or gnutls, NOT nss (web.c:401)

Jan  8 14:10:28 transmission-daemon[22386]: web NB: invalid certs will show up as 'Could not connect to tracker' like many other errors (web.c:402)
```

ryzhovau, 2019-01-10:
Looks like transmission bug/feature.

closed 2019-01-10 by ryzhovau

krehman, 2019-01-10:
Anyway to resolve this on the main transmission-daemon-openssl branch? As a test I installed it on a stand-alone Ubuntu machine and that worked. 
Thanks

zyxmon, 2019-01-10:
@krehman - here is the bug https://github.com/transmission/transmission/issues/567

You need to run transmission-daemon with environment variable CURL_CA_BUNDLE=/opt/etc/ssl/certs/ca-certificates.crt or with TR_CURL_SSL_NO_VERIFY=1. This is a log message only, it does not effect transmission behaviour,

zyxmon, 2019-01-10:
Here is the code from transmission - https://github.com/transmission/transmission/blob/master/libtransmission/web.c#L383

Here is some discussion of transmission sources regarding this problem - https://github.com/transmission/transmission/commit/44411d719c0bd6c17a794ab03eb8148f20a79b81#diff-a9ffa3cdcb7f1da80769c797a5af2c83

krehman, 2019-01-10:
@zyxmon thank you. I'll try. Trivial question - how do you add the `env` variable, `CURL_CA_BUNDLE=/opt/etc/ssl/certs/ca-certificates.crt`? Just append it in the `S88Transmission` file?

zyxmon, 2019-01-11:
Just add it before .... rc.func

williamthrilliam, 2022-12-01:
> @krehman - here is the bug [transmission/transmission#567](https://github.com/transmission/transmission/issues/567)
> 
> You need to run transmission-daemon with environment variable CURL_CA_BUNDLE=/opt/etc/ssl/certs/ca-certificates.crt or with TR_CURL_SSL_NO_VERIFY=1. This is a log message only, it does not effect transmission behaviour,

Where exactly in ubuntu do I add this? Currently I'm adding to /lib/systemd/system/transmission-daemon.service to no effect.