Entware/Entware #217: opkg update failed

giveup, 2019-01-12:
curl -v http://bin.entware.net/armv7sf-k2.6/Packages.gz
*   Trying 81.4.123.217...
* TCP_NODELAY set
* Connected to bin.entware.net (81.4.123.217) port 80 (#0)
> GET /armv7sf-k2.6/Packages.gz HTTP/1.1
> Host: bin.entware.net
> User-Agent: curl/7.54.0
> Accept: */*
> 
< HTTP/1.1 302 Found
< Connection: close
< Location: http://120.198.248.30/cache/bin.entware.net/armv7sf-k2.6/Packages.gz?ich_args2=1031-13011911048953_cc0016470f0ac8cc269dd15a279bbe4c_10001002_9c89602bd4c2f2d49039518939a83798_691887a4b7d631eb7a7e57f9e984da4b
< 
* Closing connection 0
----
PING 120.198.248.30 (120.198.248.30): 56 data bytes
64 bytes from 120.198.248.30: icmp_seq=0 ttl=56 time=20.126 ms
64 bytes from 120.198.248.30: icmp_seq=1 ttl=56 time=19.307 ms
64 bytes from 120.198.248.30: icmp_seq=2 ttl=56 time=18.249 ms

zyxmon, 2019-01-12:
Your internet connection goes through caching proxy - use another one.

giveup, 2019-01-13:
I don't use any proxy. its a DNS hijack.
Thank you for your attention.

closed 2019-01-13 by giveup