Entware/Entware #220: entware_install.sh bug

Evidlo, 2019-01-15:
`entware_install.sh` seems to be absent from this repo, so I can't make a PR.

Line 77 and 78 in `entware_install.sh` don't give the absolute path to `opkg`.  This is before the user has a chance to modify their PATH, so this always fails.

zyxmon, 2019-01-15:
Entware installation scripts have different file names - https://github.com/Entware/Entware/wiki/Alternative-install-vs-standard
Which one has a bug?

Evidlo, 2019-01-15:
I was referring to http://pkg.entware.net/binaries/armv7/installer/entware_install.sh

Maybe this is an old installer?  It was last updated about 6 months ago.

zyxmon, 2019-01-15:
It is old Entware-ng installer. Entware-ng is not supported and updated.

closed 2019-01-15 by zyxmon