Entware/Entware #226: Package request: v2ray

OpportunityLiu, 2019-01-25:
Package request
---
https://github.com/v2ray/v2ray-core
A platform for building proxies to bypass network restrictions. 

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my DD-WRT Netgear R6300V2 and get a feedback about it.

---

OpportunityLiu, 2019-01-25:
For reference, I'm using following init script 

`/opt/etc/init.d/S23v2ray`
``` sh
#!/bin/sh

ENABLED=yes
PROCS=v2ray
ARGS="-config /opt/etc/v2ray/config.json"
PREARGS=""
DESC=$PROCS
POSTCMD="sleep 1"
PATH=/opt/sbin:/opt/bin:/opt/usr/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

export V2RAY_LOCATION_ASSET=/opt/etc/v2ray
export V2RAY_LOCATION_CONFIG=/opt/etc/v2ray
export V2RAY_BUF_READV=auto
export V2RAY_RAY_BUFFER_SIZE=2

[ -z "$(which $PROCS)" ] && exit 0

. /opt/etc/init.d/rc.func

```

And postion of other files
```
opt/
  etc/
    v2ray/
      config.json
      geoip.dat
      geosite.dat
  bin/
    v2ray
    v2ctl

```

ryzhovau, 2019-01-26:
Not sure I want to add another restricrion bypass proxy.

https://github.com/Entware/Entware-ng/issues/736

linxucc, 2021-08-10:
Well @ryzhovau the need for such tool is high, and it's a very practical use on embedded devices such as routers. Which in my view makes every sense.

To be honest, openwrt and Entware was and still is for routers mainly, you don't find too many uses of routers these days. I don't get it why you think the proxy softwares are useless. It's one of the few use-cases that actually have a practical real-world value. It's where an open router really "SHINES". 

Other popular "usages" like making the router an NAS or a media server is more of a bonus hack than a real world need. If you are serious about NAS or media server and true to yourself, you'll finally agree that it's better off to some old PCs or at least get a Pi. It will save you a lot of unnecessary trouble and a lot of time, and in most cases, you'll end up with a lot better performance and stability. 

A transparent proxy is different, it's the only exclusive thing that only a router could do best, while all the other solutions are just not as good.

I have read all your replies in other related issues. I think you may have some great misunderstandings about the need for the proxy software. From what you suggested, it's likely that you think all those people in China who needs a VPN or proxy are "outlaws" and they need a proxy to do some bad business which the government prohibits. And that's something you don't want to mess up with because of the possible involvement from the law enforcement in China. But in China there's no law in regarding to the block of foreign websites. If you check the record, you'll find the government never admits such thing. There's no standard, no procedure, nor the list of which website should be blocked and which should not. 

Most people need a VPN or proxy simply for the NORMAL internet access. It's not for anything tricky, not even for porn. You could easily get all kinds of porn without a proxy in China. 
What I mean by NORMAL is that you can visit GitHub without interruptions, you could "brew install", "apt upgrade" successfully without error, you could Google a solution or watch a Youtube video. That's what people in China use proxy mainly for. Without a proxy, such basic package management commands only succeeds once out of 5 times. Github page will be OK this minute but hangs up there the next. So don't be so harsh on people asking these features, they just want a normal internet for all their devices.

So I'd say if it's not very troublesome, it will be great if those packages are officially available. It will make the router shine and introduce more people into this area. I think it's good for the community and the completeness of Entware repo. If you take a look at those projects, shadowsocks, v2ray and later xray, they all have over 20K stars in GitHub over a short time. I believe that's more than many packages that we thought "important" and got supported. The stars are not everything but that speaks something. 

And also I have an idea, if such thing involves too much work, or against your value, is it possible to add a feature like the "tap" in "homebrew", which enables 3rd-party sources to be added in the package management system??? You can "tap" another un-official source and then "brew install".

In this manner it will be a lot easier for you to maintain a centralized filtered stable core package trunk, and still have the flexibility for other people to share packages. In this case I believe a lot of people including me will be more than happy to share the load ourself. I'm no expert in package management so forgive me if anything of this is stupid.

ryzhovau, 2021-08-10:
OK then. I'll provide default config files in `/opt/etc/v2ray` which should be modified _before_ running `v2ray`. Please, provide a feedback.

closed 2021-08-10 by ryzhovau

chuxubank, 2022-05-14:
Could anyone tell why v2ray is deleted 😭

zyxmon, 2022-05-15:
We have upgraded go to 1.18 and v2ray 4.44 was not compatible with it.
Now v2ray 4.45 can be built with go 1.18 and we have restored it. Binaries will arrive soon.

chuxubank, 2022-05-15:
@zyxmon Thanks for the clarification! I'm waiting for that :)

ryzhovau, 2022-05-15:
v2ray restored thanks to @zyxmon.

chuxubank, 2022-05-15:
Thanks! Really fast!

emranovin, 2022-12-06:
Can anyone give me the link to how to set up the v2ray client on my router? (Any tutorial even in Chinese would be appreciated)
I installed the v2ray package with Entware and config the JSON but no success. I guess I should setup a SOCKS to tunnel all router's traffic through the inbounds object. Am I right?

hellkaim, 2023-07-26:
> Can anyone give me the link to how to set up the v2ray client on my router? (Any tutorial even in Chinese would be appreciated) 
This [has lot's of examples](https://github.com/XTLS/Xray-examples/tree/main)  and you SHOULD study [that man](https://xtls.github.io/config/transport.html#streamsettingsobject) 
>I installed the v2ray package with Entware and config the JSON but no success. I guess I should set up a SOCKS to tunnel all router's traffic through the inbounds object. Am I right?
It depends on what you want to achieve. You also need a server, not only client.

>If I'm right how should I do this on my AsusWrtMerlin ? (BTW why I can't find any document for the v2ray package?? I get 'Illegal instruction' error and there is no MAN or --help)
Well again it depends. but most likely with iptables. IDNK by the way.

zyxmon, 2023-07-26:
On arm routers without hard float support one should install v2ray_nohf (not v2ray).