Entware/Entware #227: Package request: tinyb

dfmt, 2019-01-26:
Package request
---
Tinyb library is used as a transportation layer for IOT/homeautomation applications such as OpenHab in order to access the Bluetooth USB Dongles directly.
"TinyB exposes the BLE GATT API for C++, Java and other languages, using BlueZ over DBus. ".
For the moment it is compatible up to Bluez 5.47

- [ ] I'm ready to test added package on my x86_64 and get a feedback about it.

More infos: https://github.com/intel-iot-devkit/tinyb

ryzhovau, 2019-01-27:
We don't keep library with no any linked application, sorry.
How you going to use it?

dfmt, 2019-01-28:
Hello, 
The tinyb library provides transportation for C and Java apps in order to be used by IOT applications such as eclipse and openhab.
There is a binding in eclipse market 
https://github.com/sputnikdev/bluetooth-manager-tinyb#prerequisites
that uses the compiled lib to access a Bluetooth dongle with BLUEZ driver in linux based SO.
This binding is intended to be used by various applications based in Java that implement the same eclipse mechanism.

ryzhovau, 2019-01-28:
Thanks for explanation.
Still not ready to provide libs without applications, linked against it. Try to compile necessary library, using our wiki.

closed 2019-01-28 by ryzhovau