Entware/Entware #229: armv7sf-k2.6 installation error

earthGavinLee, 2019-01-28:
wget -O - http://bin.entware.net/armv7sf-k2.6/installer/alternative.sh | sh

Downloading http://bin.entware.net/armv7sf-k2.6/terminfo_6.1-1_armv7-2.6.ipk
Installing locales (2.23-8) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/locales_2.23-8_armv7-2.6.ipk
sh: /opt/tmp/opkg-WISJNz/locales-tLONSy/preinst: No such file or directory
Configuring terminfo.
sh: //opt/lib/opkg/info/terminfo.postinst: No such file or directory
Configuring zoneinfo-europe.
Configuring zoneinfo-asia.
Configuring libstdcpp.
Configuring entware-release.
Configuring findutils.
Collected errors:
 * pkg_run_script: package "locales" preinst script returned status 1.
 * preinst_configure: Aborting installation of locales.
 * opkg_install_cmd: Cannot install package entware-opt.
 * pkg_run_script: package "terminfo" postinst script returned status 1.
 * opkg_configure: terminfo.postinst returned 1.
cp: /opt/etc/passwd.1: No such file or directory
cp: /opt/etc/group.1: No such file or directory
cp: /opt/etc/shells.1: No such file or directory
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to your PATH variable
Info: Add '/opt/etc/init.d/rc.unslung start' to firmware startup script for Entware services to start

This is an alternative Entware installation. We recomend to install and setup Entware version of ssh server
and use it instead of a firmware supplied one. You can install dropbear or openssh as an ssh server

earthGavinLee, 2019-01-28:
fix ncurses/entware-opt/locales/entware-upgrade :
```
diff --git a/package/libs/ncurses/Makefile b/package/libs/ncurses/Makefile
index 5a54178..34c2d68 100644
--- a/package/libs/ncurses/Makefile
+++ b/package/libs/ncurses/Makefile
@@ -154,7 +154,7 @@ endif
 endef
 
 define Package/terminfo/postinst
-#!/bin/sh
+#!/opt/bin/busybox sh
 
 grep -q "^export TERMINFO" /opt/etc/profile || echo "export TERMINFO=/opt/share/terminfo" >> /opt/etc/profile
 endef

diff --git a/entware-opt/Makefile b/entware-opt/Makefile
index 7667d77..769cdd3 100644
--- a/entware-opt/Makefile
+++ b/entware-opt/Makefile
@@ -68,7 +68,7 @@ endef
 
 # make symbolic links if busybox is not installed (standard=generic install)
 define Package/entware-opt/postinst
-#!/bin/sh
+#!/opt/bin/busybox sh
 
 if [ -e /opt/bin/busybox ]; then
     exit 0
diff --git a/entware-upgrade/Makefile b/entware-upgrade/Makefile
index 66c3751..fae2880 100644
--- a/entware-upgrade/Makefile
+++ b/entware-upgrade/Makefile
@@ -44,7 +44,7 @@ endef
 
 ifeq ($(CONFIG_TARGET_armv7soft),y)
 define Package/entware-upgrade/postinst
-#!/bin/sh
+#!/opt/bin/busybox sh
 if [[ -f /sbin/ifconfig && ! -f /opt/sbin/ifconfig ]]; then
     ln -s /sbin/ifconfig /opt/sbin/ifconfig
 fi
@@ -91,7 +91,7 @@ fi
 endef
 else
 define Package/entware-upgrade/postinst
-#!/bin/sh
+#!/opt/bin/busybox sh
 echo "Upgrade operations are not required"
 endef
 endif
diff --git a/locales/Makefile b/locales/Makefile
index b173ab2..9b6fb11 100644
--- a/locales/Makefile
+++ b/locales/Makefile
@@ -26,7 +26,7 @@ LVERSION1:=$(call qstrip,$(CONFIG_GLIBC_VERSION))
 LVERSION2:=$(subst .,,$(LVERSION1))
 
 define Package/locales/preinst
-#!/bin/sh
+#!/opt/bin/busybox sh
 
 if [[ -x "/opt/bin/locale.new" ]]; then
        /opt/bin/locale.new -V | grep $(LVERSION1) > /dev/null
@@ -40,7 +40,7 @@ endef
 
 
 define Package/locales/postinst
-#!/bin/sh
+#!/opt/bin/busybox sh
 
 echo "Entware uses separate locale-archive file independent from main system"
 if ! [ -f '/opt/usr/lib/locale/locale-archive' ]
```
fix bash:
```
diff --git a/utils/bash/Makefile b/utils/bash/Makefile
index 9b459a3..617bcde 100644
--- a/utils/bash/Makefile
+++ b/utils/bash/Makefile
@@ -74,7 +74,7 @@ CONFIGURE_ARGS+= \
        --disable-rpath \
 
 #define Package/bash/postinst
-##!/bin/sh
+##!/opt/bin/busybox sh
 #grep -q bash "$${IPKG_INSTROOT}/etc/shells" || {
 #      echo /bin/bash >> "$${IPKG_INSTROOT}/etc/shells"
 #      echo /bin/rbash >> "$${IPKG_INSTROOT}/etc/shells"

```

zyxmon, 2019-01-28:
You can suggest your fixes as a PR. Please note - they should be valid for all installations, not only for alternative one.
Entware is different from Openwrt. Some recipes for Openwrt (bash/postinst) are not valid in all cases.

closed 2019-01-28 by zyxmon