Entware/Entware #232: Package request: aha

ReneMuetti, 2019-02-01:
Package request
---
**aha** takes SGR-colored Input and prints W3C conform HTML-Code.
**aha** reads the Input from a file or stdin and writes HTML-Code to stdout.

https://manpages.ubuntu.com/manpages/cosmic/en/man1/aha.1.html

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {x86_64} and get a feedback about it.

ryzhovau, 2019-02-01:
Does it support unicode?

ReneMuetti, 2019-02-11:
https://github.com/theZiz/aha
This is the original repo.
It seems as if it is unicode.

ReneMuetti, 2019-02-11:
https://github.com/theZiz/aha/issues/62

closed 2019-05-07 by ryzhovau