Entware/Entware #233: NPM not running on ARMv7

jamiesarahg, 2019-02-04:
Hi, 

I'm trying to install node-npm. I can install node successfully, but when I install npm, even if it runs successfully, when I check for the version number, it doesn't run. I then get "Illegal Instruction" as seen below. How can I install npm. I know using node-legacy would be an option, but that requires use to use node 0.48.something and I need to use node at at least v 4.x.x.

Any suggestion on how to get node npm working?
Thank you!
``
paul@RT-AC88U-A8E8:/# opkg install node
Installing node (v8.12.0-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/node_v8.12.0-1_armv7-2.6.ipk
Configuring node.
paul@RT-AC88U-A8E8:/# opkg install node-npm
Installing node-npm (v8.12.0-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/node-npm_v8.12.0-1_armv7-2.6.ipk
Configuring node-npm.
paul@RT-AC88U-A8E8:/# npm -v
Illegal instruction
paul@RT-AC88U-A8E8:/# node-npm -v
-sh: node-npm: not found
paul@RT-AC88U-A8E8:/# node -v
v8.12.0
paul@RT-AC88U-A8E8:/# 
``

zyxmon, 2019-02-04:
You have Broadcom arm cpu and you need to install node_legacy package, not node package.
```
opkg remove --autoremove node
opkg install node_legacy

```

closed 2019-02-04 by zyxmon

jamiesarahg, 2019-02-04:
What if I need a newer version of node. To run what I need to run, it must use node 4 or higher.

zyxmon, 2019-02-04:
Your hardware does not support floating point operations. This is the last version of node, that can be compiled for it.