Entware/Entware #235: Change order of operations in syslog-ng.conf

cynicastic, 2019-02-09:
In the current /opt/etc/syslog-ng.conf file, additional customizations are loaded after everything is written to messages.  In many cases, it's advantageous to prevent certain high-volume logs from being written to the overall log, however this can't be accomplished with the current setup.

By moving the '@include' line currently at the end of syslog-ng.conf to just before the log entry, the 'flags(final)' command can be used to prevent certain logs from being written to the messages log.  Anything without the 'flags(final)' command would still be written.  Diff output:

```
--- syslog-ng.conf
+++ syslog-ng.conf.new
@@ -29,13 +29,13 @@
        file("/opt/var/log/messages");
 };

+# put any customization files in this directory
+@include "/opt/etc/syslog-ng.d/"
+
 log {
        source(src);
        source(net);
         source(kernel);
        destination(messages);
 };
-
-# put any customization files in this directory
-@include "/opt/etc/syslog-ng.d/"
```

ryzhovau, 2019-02-10:
This package came from [upstream](https://github.com/openwrt/packages/blob/master/admin/syslog-ng/files/syslog-ng.conf) with no changes. Please, send this issue to OpenWrt guys and fixed package will arrive here after next sync.

closed 2019-02-10 by ryzhovau

cynicastic, 2019-02-10:
Thanks! I'm kinda new a this (obviously) and didn't manage to figure out where it came from.  I'll ping them.