Entware/Entware #243: nftables 0.9.0_1 package is broken

token0, 2019-03-03:
```
# uname -a
Linux 3.4.113 #1 SMP Sun Jun 3 14:43:52 MSK 2018 mips GNU/Linux
# opkg update
Downloading http://bin.entware.net/mipselsf-k3.4/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
# which nft
# opkg install nftables
Installing nftables (0.9.0-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/nftables_0.9.0-1_mipsel-3.4.ipk
Configuring nftables.
# which nft
/opt/sbin/nft
# nft
/opt/sbin/nft: line 225: /home/ryzhovau/Entware/staging_dir/target-mipsel_mips32r2_glibc-2.27/../host/b in/sed: not found
/opt/sbin/nft: line 225: /home/ryzhovau/Entware/staging_dir/target-mipsel_mips32r2_glibc-2.27/../host/b in/sed: not found
/opt/sbin/nft: cd: line 1: can't cd to /home/ryzhovau/Entware/build_dir/target-mipsel_mips32r2_glibc-2. 27/nftables-0.9.0/src
```
nft is actually a shell script, and not ELF binary
[nft.txt](https://github.com/Entware/Entware/files/2923816/nft.txt)
```
#! /bin/bash

# nft - temporary wrapper script for .libs/nft
# Generated by libtool (GNU libtool) 2.4
#
# The nft program cannot be directly executed until all the libtool
# libraries that it depends on are installed.
#
# This wrapper script should never be moved out of the build directory.
# If it is, it will not operate correctly.

# Sed substitution that helps us do robust quoting.  It backslashifies
# metacharacters that are still active within double-quoted strings.
… etc
```

ryzhovau, 2019-05-08:
Can't reproduce it.
```
~  # opkg install nftables
Installing nftables (0.9.0-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/nftables_0.9.0-1_mipsel-3.4.ipk
Installing libnftnl (1.1.1-2) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/libnftnl_1.1.1-2_mipsel-3.4.ipk
Configuring libnftnl.
Configuring nftables.
~ # which nft
/opt/sbin/nft
~ # nft
nft: no command specified
~ # file /opt/sbin/nft
/opt/sbin/nft: ELF 32-bit LSB executable, MIPS, MIPS32 rel2 version 1 (SYSV), dynamically linked, interpreter /opt/lib/ld.so.1, for GNU/Linux 3.2.0, stripped
```

closed 2019-05-08 by ryzhovau

token0, 2019-05-08:
And now it works for me too!
PS: I haven't even rebooted since got that bug.

```
 # opkg update && opkg install nftables
Downloading http://bin.entware.net/mipselsf-k3.4/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
Installing nftables (0.9.0-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/nftables_0.9.0-1_mipsel-3.4.ipk
Configuring nftables.

 # nft --help
Usage: nft [ options ] [ cmds... ]

Options:
  -h, --help                    Show this help
  -v, --version                 Show version information

  -c, --check                   Check commands validity without actually applying the changes.
  -f, --file <filename>         Read input from <filename>
  -i, --interactive             Read input from interactive CLI

  -j, --json                    Format output in JSON
  -n, --numeric                 When specified once, show network addresses numerically (default behaviour).
                                Specify twice to also show Internet services (port numbers) numerically.
                                Specify three times to also show protocols, user IDs, and group IDs numerically.
  -s, --stateless               Omit stateful information of ruleset.
  -N                            Translate IP addresses to names.
  -a, --handle                  Output rule handle.
  -e, --echo                    Echo what has been added, inserted or replaced.
  -I, --includepath <directory> Add <directory> to the paths searched for include files. Default is: /opt/etc
  --debug <level [,level...]>   Specify debugging level (scanner, parser, eval, netlink, mnl, proto-ctx, segtree, all)

# which nft
/opt/sbin/nft

# uptime
 00:04:00 up 181 days, 14:16
```