Entware/Entware #247: Issue after adding ABI_VERSION to package name

boretom, 2019-03-12:
Hi,

To get to know the Entware build system I created a package for nmon, which does depend on ncurses.
It and its dependencies compiled fine. And I set up my own binary feed for this package to install it on my QNAP NAS ([my bin repo](https://entware.kupper.org), [my github repo](https://github.com/boretom/boretom-entware)).

The installation of nmon failed because it wanted to install libncurses6 library but libncurses was already installed and provided the same libraries.

Trying to find the root of it I came across an [OpenWrt commit from Jan 17](https://github.com/openwrt/openwrt/commit/e6bcf1e4acc4901c1803cdccc4b9fabfd6d84c90) which added the ABI_VERSION to the package name.

That caused the trouble since Entware merged these changes but the packages of the binary repo did not get recompiled and
But since I used the latest commit of Entware to compile the package the Packages index referred to libncurses6, not libncurses.

So my question is, do you have any plans to recompile all packages to have the ABI_VERSION included in the package name?

Of course I can easily fix this myself for myself by disabling ABI_VERSION in the ncurses library. Still I'd think it would be great if you could recompile all packages at some time to be consistent.

Thanks for a great (and much simpler than Optware) building system
/Thomas

zyxmon, 2019-03-12:
Yes. We are in the process of recompiling and fixing packages now.

boretom, 2019-03-12:
Great thank you.

closed 2019-03-12 by boretom

ryzhovau, 2019-03-13:
> So my question is, do you have any plans to recompile all packages to have the ABI_VERSION included in the package name?

This solution is quite questionable and I'm not sure we will take it. It helps to recompile package, dependency on library with ABI_VERSION change, but it useless to end user. This collision will stay on your side:

> The installation of nmon failed because it wanted to install libncurses6 library but libncurses was already installed and provided the same libraries.

Because `opkg` can't handle ABI change at all. Implementing ABI_VERSION will cause ["data clashes error"](https://github.com/Entware/Entware/wiki/Home/_compare/c268a2996ae1b1ce6e54eb3dca5fc90cae038354...e8809cc1b2db383074f33b3ad1eb6d8d024962a2) for 50+ libraries at this moment.

boretom, 2019-03-13:
Hi Alexander,

Since you guys are much more into that topic I assume you're right. So you
will revert that specific OpenWrt commit and remove the ABI_VERSION?

I'm happy with whatever solutions makes it possible for me to add the my
custom repo and still use the official one.

ryzhovau, 2019-03-13:
Yes, mostly:)
We've got few days before our major update finished, so we going to wait this few days for (a miracle?) OpenWrt guys to fix their `opkg`.