Entware/Entware #249: Unsupported CPU architecture - QNAP TS-259 Pro+ / Intel Atom D525

trinkleinj, 2019-03-15:
I'm attempting to install Entware on a QNAP TS-259 Pro+ with an Intel Atom D525. It is supposed to be an x86_64 processor, but both through the WebUI and the CLI, installation fails with "Unsupported CPU architecture". Contrary to what it claims, entware is *not* currently installed.

sh Entware_1.00std.qpkg
Install QNAP package on TS-NAS...
33+1 records in
33+1 records out
./
./qinstall.sh
./package_routines
./qpkg.cfg
0+1 records in
0+1 records out
3+1 records in
3+1 records out
Entware  is already installed. Setup will now perform package upgrading.
Entware 1.00 installation failed. Unsupported CPU architecture, quitting installation of Entware
Installation Abort.

zyxmon, 2019-03-15:
Please run 'uname -a'. Probably your nas does not support 64bit instructions. You can install entware-ng in this case.

trinkleinj, 2019-03-15:
uname -a gives me:
Linux moserver 3.4.6 #1 SMP Fri Oct 26 15:33:41 CST 2018 i686 unknown

trinkleinj, 2019-03-16:
Installing Entware-ng was successful. I guess the system software is 32 bit despite the 64 CPU. Thanks!

closed 2019-03-16 by trinkleinj