Entware/Entware #250: Package Request:  Bleachbit

rosede12, 2019-03-20:
Package request
---
Please add bleachbit or similar to the application list.  This is needed to securely wipe files and/or directory to ensure security and integrity of devices. 

- [X ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2019-03-24:
It's a GTK application.
It's not suitable for embedded platforms.

closed 2019-03-24 by ryzhovau

rivy, 2019-04-29:
@ryzhovau , why do believe that GTK applications are unsuitable?

dia3olik, 2019-09-07:
at least you could add secure-delete which is command line