Entware/Entware #251: Bug report: opkg install wget (aarch64-k3.10)

JACK-THINK, 2019-03-24:
Bug report (aarch64-k3.10):

```
opkg install wget
```
return

```
Collected errors:
 * opkg_download: Failed to download http://bin.entware.net/aarch64-k3.10/wget_1.20.1-3_aarch64-3.10.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download wget. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package wget.
```

The correct file name is `wget_1.20.1-3a_aarch64-3.10.ipk`

NitroOxid, 2019-03-24:
```
/opt/home/admin/ipk # opkg install wget
Unknown package 'wget'.
Collected errors:
 * opkg_install_cmd: Cannot install package wget.
```
The architecture of mipsel is almost the same problem.
I can't install more than half of python packages, I can't install wget, youtube-dl, dav2fs, and etc

Packages and dependencies can be installed manually by downloading ipk from the repository.

zyxmon, 2019-03-24:
Openwrt and hense Entware updated openssl from 1.0.2 to 1.1.1.
wget depends on openssl and is not installed by default. wget is used to install packages (default is wget applet from busybox). When you run update - openssl is updated and wget stops working. (busybox applet does not have problem).
You'll need to manually to restore wget (busybox) with `opkg remove wget; ln -s  /opt/bin/busybox /opt/bin/wget`
Some other packages may also need some manual operations.

closed 2019-03-24 by zyxmon

JACK-THINK, 2019-03-24:
@zyxmon 
Apparently you didn't get my point. I install (not update) Entware freshly and then use `opkg install wget` to install wget. Then it gives me information as following
```
Collected errors:
 * opkg_download: Failed to download http://bin.entware.net/aarch64-k3.10/wget_1.20.1-3_aarch64-3.10.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download wget. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package wget.
```

Apparently, the reason why opkg failed to download http://bin.entware.net/aarch64-k3.10/wget_1.20.1-3_aarch64-3.10.ipk is that the actual name of the package is `wget_1.20.1-3a_aarch64-3.10.ipk`. There is **an extra 'a' following the version number** in the name of package.

zyxmon, 2019-03-25:
Cannot confirm
```
 # opkg update
Downloading http://bin.entware.net/x64-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
 # opkg install wget
Upgrading wget on root from 1.20-1a to 1.20.1-3a...
Downloading http://bin.entware.net/x64-k3.2/wget_1.20.1-3a_x64-3.2.ipk
libpcre2 was autoinstalled and is now orphaned, removing.
Removing package libpcre2 from root...
Configuring wget.

```
It looks like your package index has previous version of wget that was updated yesterday.

ryzhovau, 2019-03-25:
Please, try again, that was indexing bug on our side.